rwaldron/johnny-five #1593: Play wav files

atrauzzi, 2019-08-11:
I've been using Circuit Python on my Circuit Playground Express and while I'm able to get most functionality, my device is always going to be connected to something bigger (PC, Rasberry Pi).

That means I can potentially use J5 to control my device.  Unfortunately, one thing I haven't been able to figure out yet is how to play a 16bit PCM wav file in J5 that I'm able to play from Circuit Python.

Would be super nice if J5 had a way to play very simple wav files out the piezo speaker.

dtex, 2019-08-12:
The short answer is you can’t, but the true answer is you can but you have to do some work. Enough work in fact that it may not be practical.  Depends on the platform. You mentioned there will always be a bigger PC connected to the microcontroller. Is there any reason you can’t just play the sound from the computer?

atrauzzi, 2019-08-12:
Sadly the computer won't have speakers attached and... This might sound odd... But I'd like to use the tiny low fi speaker for my application. 😃

dtex, 2019-08-13:
So what platform are we talking about?

atrauzzi, 2019-08-13:
Well I'd be plugging a circuit playground express into a raspberry Pi and using it headless.

rwaldron, 2019-08-13:
I spent some time earlier today trying to figure out where the code for their `cpx.play_file(...)` function is and didn't have any luck. There seems to be a soup of python libraries. I'm sure if I had spent a little more time, I'd have found what I was looking for.

dtex, 2019-08-14:
With Johnny-Five running on the Pi and standardFirmataPlus on the Circuit Playground Express (CPE) you won't be able to play a wav file that lives on the Pi through the piezo on your CPE. It's just not possible through the firmata protocol. 

One possibility (that pushes the limit of practicality) would be to create a custom version of firmata that can receive a SYSEX command to play wav files through the piezo from the sd card on the CPE using the same techniques as cpx.play_file (which I couldn't find either). You can see other examples of custom firmata features at http://firmatabuilder.com/ under "Contributed Features".

Another possibility (that I like a little better) would be to build an I2C backpack that can have a custom sketch (that includes the song/sound clips) that plays the sound over an attached piezo, but then you wouldn't be using the built-in piezo of the CPE. On the other hand, you will have built something that can be used on any platform that supports I2C (which is a lot).

Either of those paths will require a quite a bit of Arduino C programming.

I doubt we will find someone else with a suitable combination of time and motivation to take this on, but if you are interested in either of those paths I will support you with advice, answers to questions and pointers to the right experts as best I can.

Should I leave this ticket open for you (so we can follow along) or should I close it?

atrauzzi, 2019-08-15:
Well, I should mention, I'm not opposed to putting the wav file on the device. That's what I do with Circuit Python.  Don't know if that changes or improves things. It's like 33kb 😄

atrauzzi, 2019-08-15:
My preferred approach would be to not require a backpack though as the device already has a speaker and I wouldn't want any additional size or weight.

Regarding your question, I don't really have the time or comfort with C and low level device coding to get my hands in this. I'm a high level programmer, so delving into these concepts is more than likely above my skill level.

closed 2020-08-16 by stale[bot]