rwaldron/johnny-five #1596: Update examples / documentation

dtex, 2019-08-15:
Stop using this, use arrow functions where appropriate, spread operators, const on requires, and appropriate variable scoping (let).

Here is a recent example:

https://github.com/rwaldron/johnny-five/blob/master/eg/esc-keypress.js

- [ ] accelerometer.js
- [ ] altimeter.js
- [ ] animation.js (@dtex)
- [ ] barometer.js
- [ ] board.js
- [ ] board.options.js
- [ ] board.pins.js
- [ ] button.js (@monteslu)
- [ ] color.js
- [ ] compass.js
- [ ] datasheets.md
- [ ] esc.js (@HipsterBrown)
- [ ] evshield.js
- [ ] expander.js
- [ ] fn.js
- [ ] gps.js (@dtex)
- [ ] gripper.js
- [ ] gyro.js
- [ ] hygrometer.js
- [ ] imu.js
- [ ] johnny-five.js
- [ ] joystick.js
- [ ] keypad.js
- [ ] lcd-chars.js
- [ ] lcd.js
- [ ] light.js (@reconbot)
- [ ] motion.js
- [ ] motor.js (@dtex)
- [ ] orientation.js
- [ ] piezo.js
- [ ] pin.js
- [ ] ping.js
- [ ] proximity.js
- [ ] reflectancearray.js
- [ ] relay.js
- [ ] repl.js
- [ ] sensor.js (@ajfisher)
- [ ] servo.js (@dtex)
- [ ] shiftregister.js
- [ ] sleep.js
- [ ] sonar.js
- [ ] stepper.js
- [ ] switch.js
- [ ] thermometer.js
- [ ] wii.js
- [ ] digits.js
- [ ] led-chars.js
- [ ] led.js (@ryanjgill2)
- [ ] ledcontrol.js
- [ ] leds.js
- [ ] matrix.js
- [ ] rgb.js
- [ ] rgbs.js

ryanjgill, 2019-08-17:
Can I combine a few of the led examples in to a single push or should I just do them all as separate PR's? I was thinking about doing some of the led sub sections together like `blink`, `pulse` and `pulse with animation` as one.

dtex, 2019-08-17:
Grouped together is fine. It might be nice to have each group of examples (by class) as a separate commit, JIC.

reconbot, 2019-08-18:
Are object shorthands ok?

eg

from
```js
const foo = {
   read: function() {
      //
   }
}
```

to
```js
const foo = {
   read() {
      //
   }
}
```

reconbot, 2019-08-18:
I checked the object shorthand is aok with supported node versions. I submitted #1601 and before I realized I should have scoped down to just the docs I updated the class too. Let me know if I should take that out.

dtex, 2019-08-19:
@ryanjgill put in some shorthand properties on some LED examples as well. Let's get sign-off from @rwaldron before we do any others.

dtex, 2019-08-19:
Shorthand properties are OK

closed 2020-09-20 by stale[bot]