rwaldron/johnny-five #1614: Sensor.Digital.on('change') way slower than board.digitalRead

Pimentoso, 2019-08-27:
Board: Arduino nano
Hardware you are having an issue with: in my case, phototransistors
Version of Johnny-Five: 1.1.0
What your expectations are: Sensor.Digital.on('change') callback called as fast as possible
What the actual outcome is: on very rapid events firing, lots of callback calls are lost
Steps to reproduce (including code samples):

- connect phototransistor (in my case, a TEMT6000) to digital pin 6, 5v, and ground
- point a laser LED to it
- place a USB fan between the laser and the phototransistor (or something that can fire events at like 100hz)

Run the following code
`var five = require("johnny-five"),
  board = new five.Board(),
  i = 0;

board.on("ready", function() {
  sensor1 = new five.Sensor.Digital({pin: 6, freq: 1});
  sensor1.on("change", function(val) {
    if (val == 0) {
      console.log(i++);
    }
  });
});`
This one will show very slow readings.

`var five = require("johnny-five"),
  board = new five.Board(),
  i = 0;

board.on("ready", function() {
  this.samplingInterval(1);
  this.pinMode(6, five.Pin.INPUT);
  this.digitalRead(6, function(val) {
    if (val == 0) {
      console.log(i++);
    }
  });
});`
This one will be way faster.

Looks like there's some overhead when using the Sensor.Digital class. I haven't dived into the code a lot, but the Sensor class code fires the digital events first and then returns immediately if type == 'digital' so I don't think the problem is there.

dtex, 2019-09-06:
@Pimentoso I suspect nobody has a rig suitable for testing this so that's why no response.

Glancing at the second example, does it actually wait for changes? On the surface, it looks like it would blast increments as long as the laser is blocked.

Pimentoso, 2019-09-07:
@dtex it actually waits for changes, as per documentation

> Note: digitalRead will only call its handler when the value of the pin changes.

dtex, 2019-09-07:
Simple test rig. A 10k resistor bridging pin 6 and 3 with this code:

````js
const {Board, Led} = require("johnny-five");
const board = new Board();

const frequency = 100;
const period = 1000 / frequency;

let changes = 0;
let observedChanges = -1; // always one on intialization

board.on("ready", () => {
  board.samplingInterval(1);
  
  const led = new five.Led(3);
  
  board.pinMode(6, five.Pin.INPUT);
  board.digitalRead(6, function(val) {
    observedChanges++;
  });

  setInterval(() => { 
    led.toggle(); 
    changes++; 
  }, period);

  setTimeout (() => { 
    console.log(`${observedChanges} of ${changes} state changes observed`);
    process.exit();
  }, 10000);

});
````

dtex, 2019-09-07:
At 100hz both digitalRead and sensor change events are working perfectly for me using the above code. At 200hz the sensor change event misses a few. In both cases, changes are not emitted at the requested frequency due to either initialization time or code overhead. Still testing to figure out why.

dtex, 2019-09-07:
I should have asked more about your setup. I assumed you were using an Arduino connected to your computer. Is that right?

Pimentoso, 2019-09-08:
Yes indeed it's a cheap nano clone with micro usb, connected directly to a MBP, also tried variable length usb cables, nothing changed. I was not really expecting 1 read per millisecond even with "freq:1"  because of the usb protocol, j5 being javascript and other factors, but in my test script similar to yours (checking the number of reads in 10 seconds) I always got oscillating values, averaging 120 hz which is quite low for my need. 

Again I solved it using board.digitalRead which works fine up to 250 hz so it's fine, but I opened this issue for reference.

closed 2020-09-11 by stale[bot]