rwaldron/johnny-five #1618: Sunfounder Picar-V + johnny-five

5ocjal, 2019-09-01:
Hi guys, got a question. If anyone tried to run this Sunfounder Picar-V car with johhny-five? Is there any documentation where I can find which Raspberry's pin a component uses?

HipsterBrown, 2019-09-01:
So looking at the source code for that project (https://github.com/sunfounder/SunFounder_PiCar-V/blob/master/ball_track/ball_tracker.py), it appears to be using the PCA9685 servo controller. There is an example of using that controller with Johnny-Five in the docs -> http://johnny-five.io/examples/servo-PCA9685/ Not sure about the pin numbers for that controller. 

It also appears to be using the TB6612 motor driver for the back wheels, here is an example of that usage with Johnny-Five -> http://johnny-five.io/examples/motor-TB6612FNG/ The pins used by the project seem to be `PWM4` and `PWM5`.

closed 2019-09-01 by rwaldron

5ocjal, 2019-09-01:
Thank you rwaldron. I was able to make servo running, I'll check out the motors in a while. :)

5ocjal, 2019-09-01:
Thank you HipsterBrown. I was able to make servo running, I'll check out the motors in a while. :)

5ocjal, 2019-09-13:
One more question. I was made to run one motor with this code:

```
const {Board, Motors} = require('johnny-five');
const raspio = require('raspi-io').RaspiIO;
const invertPWM = true;

const board = new Board({
    io: new raspio()
});

board.on("ready", function() {

  var tb6612 = new Motors([
    { controller: "PCA9685", pins: { dir: 4, cdir: 5, pwm: 6 } },

  ]);
  tb6612.forward(255);
  board.wait(3000, ()=> tb6612.stop);
});
```
Unfortunately, when I execute this code, only the right motor starts and it alfo seems that this is the result of the controller works. What's more, after 3 seconds it does not stop, but it can work indefinitely, even if I stop the script with 'ctrl + c'. I tried to use the example that HipsterBrown linked, but I couldn't start the motors anyway.

Wiki from Sunfounder didn't help me much either.
--->  http://wiki.sunfounder.cc/index.php?title=TB6612_DC_Motor_Driver_Module

I will be grateful if someone looks at it.