rwaldron/johnny-five #1619: Having trouble reading from a Lidar Lite to a Raspberry Pi

thatcotter, 2019-09-06:
**Board**:  Pi Model 3B V1.2
**Sensor**: Lidar Lite v3

I've been setting up a raspberry pi B+ to read from a Lidar Lite, but keep getting an "invalid arguments" error. I have the lidar hooked up to the respective SDA and SCL pins, but haven't been able to read anything with ` i2cdetect -y -a 1`. Is there a different set of GPIO pins I should hook it up to?

```
const Raspi = require('raspi-io').RaspiIO;
const {Board, Proximity, Led} = require('johnny-five');
const board = new Board({
  io: new Raspi()
});

board.on("ready", () => {
  const lidar = new Proximity({
    controller: "LIDARLITE"
  });
  

  lidar.on("change", () => {
    console.log("Proximity: ");
    console.log("  cm  : ", proximity.cm);
    console.log("  in  : ", proximity.in);
    console.log("-----------------");
  });
});
```

**Error:**
> 1567805703383 Available Raspi IO
> 1567805703394 Connected Raspi IO
> 1567805703403 Repl Initialized
> >> /home/pi/Desktop/../node_modules/j5-io/dist/index.js:481
>             throw new Error('Invalid arguments');
>             ^
> 
> Error: Invalid arguments
>     at J5IO.i2cWrite (/home/pi/Desktop/../node_modules/j5-io/dist/index.js:481:19)
>     at Proximity.<anonymous> (/home/pi/Desktop/../node_modules/johnny-five/lib/proximity.js:286:19)
>     at Proximity.value (/home/pi/Desktop/../node_modules/johnny-five/lib/proximity.js:296:9)
>     at new Proximity (/home/pi/Desktop/../node_modules/johnny-five/lib/proximity.js:503:10)
>     at Board.board.on (/home/pi/Desktop/../index.js:19:18)
>     at Board.emit (events.js:203:15)
>     at Board.EventEmitter.emit (domain.js:448:20)
>     at process._tickCallback (internal/process/next_tick.js:61:11)
>     at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
>     at startup (internal/bootstrap/node.js:283:19)

dtex, 2019-09-06:
The particulate signature for i2cWrite used by Johnny-Five here isn't handled in j5-io.

@nebrius This is similar to the other pattern we caught that firmata.js allowed, but was undocumented. The params being passed are (Number address, Number register, Number inByte). j5-io expects an array in that third parameter. Johnny-Five probably should have used i2cWriteReg instead of i2cWrite, but to ensure compatibility with whatever else might be out there, we should update j5-io to allow for this signature. 

I'll update the firmata.js readme to include this pattern and get a PR in.

@thatcotter I'll get the PR's in by this weekend but if you want to move forward right away, edit line 286 in ```./node_modules/johnny-five/lib/proximity.js```. Change:
````js
this.io.i2cWrite(address, this.REGISTER.ENABLE, 0x04);
````
to
````js
this.io.i2cWriteReg(address, this.REGISTER.ENABLE, 0x04);
````

Let me know if that works and if it does let me know how you like that Lidar Lite. I want to try one out someday.

thatcotter, 2019-09-10:
Thanks, @dtex! I swapped out that line, and it works great now.

closed 2019-09-10 by thatcotter