rwaldron/johnny-five #1624: TB6612 Motors

5ocjal, 2019-09-14:
Hi guys, got a question. If anyone tried to run this Sunfounder Picar-V car with johhny-five?
While, servos work just fine, motors don't.

I was made to run one motor with this code:

```
const {Board, Motors} = require('johnny-five');
const raspio = require('raspi-io').RaspiIO;
const invertPWM = true;

const board = new Board({
    io: new raspio()
});

board.on("ready", function() {

  var tb6612 = new Motors([
    { controller: "PCA9685", pins: { dir: 4, cdir: 5, pwm: 6 } },

  ]);
  tb6612.forward(255);
  board.wait(3000, ()=> tb6612.stop);
});
```

Unfortunately, when I execute this code, only the right motor starts and it alfo seems that this is the result of the controller works. What's more, after 3 seconds it does not stop, but it can work indefinitely, even if I stop the script with 'ctrl + c'. 

I tried to use the example that HipsterBrown linked here, but I couldn't start the motors anyway.
---> http://johnny-five.io/examples/motor-TB6612FNG/

Wiki from Sunfounder didn't help me much either.
---> http://wiki.sunfounder.cc/index.php?title=TB6612_DC_Motor_Driver_Module

I will be grateful if someone looks at it.

dtex, 2019-09-14:
Hi @5ocjal 

I haven't done anything with the Sunfounder, but I have done a lot of work with motors and motor controllers. First, looking at the image on the Sunfounder TB6612 wiki page I can see that it does not require 3 pins to drive it. It's not unheard of for a motor controller to require 3-pins and then for the breakout manufacturer to add hardware logic that controls one of those pins so you only need two pins from your microcontroller (or Pi in this case) to drive it. 

We just need to figure out which pins on the Raspberry Pi are connected to the MA, PWMA, MB and PWMB pins on the motor controller. Then use code like:

````js
var tb6612 = new Motors([
    { controller: "PCA9685", pins: [PWMA, MA] },
    { controller: "PCA9685", pins: [PWMB, MB] }
]);
````

I'm trying to digest the Picar library code so I can see how it's wired, but I'm neither a Pi guy or a Py guy so it might take me a little while.  Can you visually tell how it's wired?

dtex, 2019-09-14:
Note: it could turn out that PWMA and PWMB are connected to the PCA9685 and MA and MB are not connected to the PCA9685 (they don't need to be). If that's the case, this is going to get tricky.

dtex, 2019-09-14:
So the reason I can't find the pins in the repo is that picar.py is not included in the repo and all the configuration details are in there :-/

5ocjal, 2019-09-14:
**dtex** - first of all - thank you for your time. 

Please, looke at this --> https://www.sunfounder.com/learn/download/X1BWQ19SYXNwYmVycnlfUGlfU21hcnRfVmlkZW9fQ2FyX1YyLjAucGRm/dispi

Here is the wiring schema (page 18-19). Problem is, its wired with something called Robot HATS not directlu to Pi.

5ocjal, 2019-09-14:
UPDATE: This Robot Hats is probably PCF8591t.

dtex, 2019-09-14:
That wiring diagram helped. PWMA = pin 4 on the PCA9685 and PWMB = pin 5.

It looks like pins MA and MB are indeed GPIO pins on the Pi (piped through the hat) so we will need a custom hybrid controller. I can help with that, but we will need to figure out which of the Pi's GPIO pins connect to MA and MB. I can see that it is these two pins on the hat:

[image]

but we need to figure out what those are connected to on the Pi.

5ocjal, 2019-09-14:
On page 64 there is a information that driver code is in this repo -->
https://github.com/sunfounder/SunFounder_PiCar

Take a look on this -->
https://github.com/sunfounder/SunFounder_PiCar/blob/master/picar/SunFounder_TB6612/TB6612.py

Code line 125+ says:

```
	print("********************************************")
	print("*                                          *")
	print("*           SunFounder TB6612              *")
	print("*                                          *")
	print("*          Connect MA to BCM17             *")
	print("*          Connect MB to BCM18             *")
	print("*         Connect PWMA to BCM27            *")
	print("*         Connect PWMB to BCM12            *")
	print("*                                          *")
	print("********************************************")
	GPIO.setmode(GPIO.BCM)
	GPIO.setup((27, 22), GPIO.OUT)
	a = GPIO.PWM(27, 60)
	b = GPIO.PWM(22, 60)
```

But I'm not sure is it any help for you.

dtex, 2019-09-14:
So, I think this oughtta do it, The digital pin writes should fall through to the default controller. Once we figure out what the MA and MB GPIO pins on the Pi are, just replace MA? and MB? with those values:

````js
const {Board, Motors} = require('johnny-five');
const raspio = require('raspi-io').RaspiIO;
const invertPWM = true;

const board = new Board({
    io: new raspio()
});

board.on("ready", function() {

  const PicarVMotorController = {
    setPWM: {
      writable: true,
      value: function(pin, speed) {
        var state = priv.get(this);
        state.expander.analogWrite(pin, speed);
      }
    },
    initialize: {
      value: function(opts) {
        var state = priv.get(this);
        this.address = opts.address || 0x40;
        this.pwmRange = opts.pwmRange || [0, 4080];
        this.frequency = opts.frequency || 50;
        state.expander = Expander.get({
          address: this.address,
          controller: this.controller,
          bus: this.bus,
          pwmRange: this.pwmRange,
          frequency: this.frequency,
        });
        this.pins.pwm = state.expander.normalize(this.pins.pwm);
      }
    }
  };

  var tb6612 = new Motors([
    { controller: "PicarVMotorController", pins: { dir: MA?, pwm: 4 }, invertPWM: true },
    { controller: "PicarVMotorController", pins: { dir: MB?, pwm: 5 }, invertPWM: true }
  ]);

  tb6612.forward(255);
  board.wait(3000, ()=> tb6612.stop);
});

````

5ocjal, 2019-09-14:
Okey, so I set MA to pin 11 and MB to pin 12 according to this: 

[image]

Run the code with 'sudo node motors.js' but got this:

```
pi@raspberrypi:~/Luke $ sudo node motors.js 
1568502228719 Available Raspi IO  
1568502228731 Connected Raspi IO  
1568502228744 Repl Initialized  
>> /home/pi/Luke/node_modules/johnny-five/lib/board.js:1019
  var requirements = controller.requirements && controller.requirements.value;
                                ^

TypeError: Cannot read property 'requirements' of undefined
    at Motor.Board.Controller (/home/pi/Luke/node_modules/johnny-five/lib/board.js:1019:33)
    at new Motor (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:571:22)
    at Motors.Collection (/home/pi/Luke/node_modules/johnny-five/lib/mixins/collection.js:53:25)
    at new Motors (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:1171:14)
    at Board.<anonymous> (/home/pi/Luke/motors.js:37:16)
    at Board.emit (events.js:203:15)
    at Board.EventEmitter.emit (domain.js:448:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
    at startup (internal/bootstrap/node.js:283:19)
```

dtex, 2019-09-14:
Oops. Sorry, I have no way to test. 

Change:

 ````js
var tb6612 = new Motors([
    { controller: "PicarVMotorController", pins: { dir: MA?, pwm: 4 }, invertPWM: true },
    { controller: "PicarVMotorController", pins: { dir: MB?, pwm: 5 }, invertPWM: true }
  ]);
````
to (remove the quotes on the controller name)

````js
 var tb6612 = new Motors([
    { controller: PicarVMotorController, pins: { dir: 11, pwm: 4 }, invertPWM: true },
    { controller: PicarVMotorController, pins: { dir: 12, pwm: 5 }, invertPWM: true }
  ]);
````

5ocjal, 2019-09-14:
Oh, now I get that one:

```
pi@raspberrypi:~/Luke $ sudo node motors.js 
1568503579070 Available Raspi IO  
1568503579086 Connected Raspi IO  
1568503579100 Repl Initialized  
>> /home/pi/Luke/motors.js:20
        var state = priv.get(this);
                    ^

ReferenceError: priv is not defined
    at Motor.value (/home/pi/Luke/motors.js:20:21)
    at new Motor (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:623:10)
    at Motors.Collection (/home/pi/Luke/node_modules/johnny-five/lib/mixins/collection.js:53:25)
    at new Motors (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:1171:14)
    at Board.<anonymous> (/home/pi/Luke/motors.js:36:16)
    at Board.emit (events.js:203:15)
    at Board.EventEmitter.emit (domain.js:448:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
    at startup (internal/bootstrap/node.js:283:19)
```

5ocjal, 2019-09-14:
Can you explain what's going on here?

```
  const PicarVMotorController = {
    setPWM: {
      writable: true,
      value: function(pin, speed) {
        var state = priv.get(this);
        state.expander.analogWrite(pin, speed);
      }
    },
    initialize: {
      value: function(opts) {
        var state = priv.get(this);
        this.address = opts.address || 0x40;
        this.pwmRange = opts.pwmRange || [0, 4080];
        this.frequency = opts.frequency || 50;
        state.expander = Expander.get({
          address: this.address,
          controller: this.controller,
          bus: this.bus,
          pwmRange: this.pwmRange,
          frequency: this.frequency,
        });
        this.pins.pwm = state.expander.normalize(this.pins.pwm);
      }
    }
```

I like to understand what is happening in the code, and here I am a little bit confused. :)

dtex, 2019-09-14:
Okay, sorry for the back and forth. 

This should fix that and one other error:

````js
const {Board, Motors} = require('johnny-five');
const raspio = require('raspi-io').RaspiIO;
const invertPWM = true;
let priv = new Map();

const board = new Board({
    io: new raspio()
});

board.on("ready", function() {

  const PicarVMotorController = {
    setPWM: {
      writable: true,
      value: function(pin, speed) {
        var state = priv.get(this);
        state.expander.analogWrite(pin, speed);
      }
    },
    initialize: {
      value: function(opts) {
        var state = priv.get(this);
        this.address = opts.address || 0x40;
        this.pwmRange = opts.pwmRange || [0, 4080];
        this.frequency = opts.frequency || 50;
        state.expander = Expander.get({
          address: this.address,
          controller: this.controller,
          bus: this.bus,
          pwmRange: this.pwmRange,
          frequency: this.frequency,
        });
        this.pins.pwm = state.expander.normalize(this.pins.pwm);
      }
    }
  };

  var tb6612 = new Motors([
    { controller: "PicarVMotorController", pins: { dir: 11, pwm: 4 }, invertPWM: true },
    { controller: "PicarVMotorController", pins: { dir: 12, pwm: 5 }, invertPWM: true }
  ]);

  tb6612.forward(255);
  board.wait(3000, ()=> tb6612.stop);
});
````

dtex, 2019-09-14:
> Can you explain what's going on here?

I'll do my best. Johnny-Five supports a wide range of motor controllers, microcontrollers, etc. To accommodate all the different motor and controller types we allow the base Motor class to be decorated with device and controller specific code. 

Here we've created a new motor controller (PicarVMotorController) that uses a PCA9685 for the PWM pin and a regular GPIO pin on the Pi (or whatever) for the DIR pin. 

The setPWM property handles writing the speed value to the correct pin on the PCA9685.

Since we haven't defined a setPin property, that falls through to the default case which is a digital pin write.

This is all probably easier to understand by examining johnny-five/lib/motor.js.

There aren't many examples out there where user code defines a motor controller, as most possible cases are already handled by Johnny-Five. If this works, I'll add it to Johnny-Five properly so others can use it more easily.

5ocjal, 2019-09-14:
Thank you once more. It's my first time with Johnny-five and Raspberry - still learning. :)

Ok, I belive now we should import and declare Expander? :)

```
pi@raspberrypi:~/Luke $ sudo node motors.js 
1568504866940 Available Raspi IO  
1568504866959 Connected Raspi IO  
1568504866981 Repl Initialized  
>> /home/pi/Luke/node_modules/johnny-five/lib/expander.js:2014
    throw new Error("Expander.get(...) expects controller name to be a string");
    ^

Error: Expander.get(...) expects controller name to be a string
    at Function.Expander.get (/home/pi/Luke/node_modules/johnny-five/lib/expander.js:2014:11)
    at Motor.value (/home/pi/Luke/motors.js:26:35)
    at new Motor (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:623:10)
    at Motors.Collection (/home/pi/Luke/node_modules/johnny-five/lib/mixins/collection.js:53:25)
    at new Motors (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:1171:14)
    at Board.<anonymous> (/home/pi/Luke/motors.js:38:16)
    at Board.emit (events.js:203:15)
    at Board.EventEmitter.emit (domain.js:448:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
```

dtex, 2019-09-15:
Yes
````js
const {Board, , Expander, Motors} = require('johnny-five');
````

5ocjal, 2019-09-15:
```
pi@raspberrypi:~/Luke $ sudo node motors.js 
1568539722770 Available Raspi IO  
1568539722781 Connected Raspi IO  
1568539722794 Repl Initialized  
>> /home/pi/Luke/node_modules/johnny-five/lib/expander.js:2014
    throw new Error("Expander.get(...) expects controller name to be a string");
    ^

Error: Expander.get(...) expects controller name to be a string
    at Function.Expander.get (/home/pi/Luke/node_modules/johnny-five/lib/expander.js:2014:11)
    at Motor.value (/home/pi/Luke/motors.js:26:35)
    at new Motor (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:623:10)
    at Motors.Collection (/home/pi/Luke/node_modules/johnny-five/lib/mixins/collection.js:53:25)
    at new Motors (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:1171:14)
    at Board.<anonymous> (/home/pi/Luke/motors.js:38:16)
    at Board.emit (events.js:203:15)
    at Board.EventEmitter.emit (domain.js:448:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
```

5ocjal, 2019-09-15:
Did not help.

```
pi@raspberrypi:~/Luke $ sudo node motors.js 
1568540408370 Available Raspi IO  
1568540408382 Connected Raspi IO  
1568540408396 Repl Initialized  
>> /home/pi/Luke/node_modules/johnny-five/lib/expander.js:2014
    throw new Error("Expander.get(...) expects controller name to be a string");
    ^

Error: Expander.get(...) expects controller name to be a string
    at Function.Expander.get (/home/pi/Luke/node_modules/johnny-five/lib/expander.js:2014:11)
    at Motor.value (/home/pi/Luke/motors.js:28:35)
    at new Motor (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:623:10)
    at Motors.Collection (/home/pi/Luke/node_modules/johnny-five/lib/mixins/collection.js:53:25)
    at new Motors (/home/pi/Luke/node_modules/johnny-five/lib/motor.js:1171:14)
    at Board.<anonymous> (/home/pi/Luke/motors.js:40:16)
    at Board.emit (events.js:203:15)
    at Board.EventEmitter.emit (domain.js:448:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
```

dtex, 2019-09-15:
Ah, I was passing in the controller name as the type of expander. I'm sorry about all this back and forth, I should really bust out the hardware if I'm going to try and program this. In the meantime, here's the fix for that:

Change
````js
controller: this.controller,
````
to
````js
controller: "PCA9685"
````

dtex, 2019-09-15:
Okay, I went ahead and busted out some hardware and I can see that a user-defined controller with an expander has a problem (I cannot at the expander instance to the private state object in the motor class). I've created [this gist](https://gist.github.com/dtex/640bd48b10244b6ef8d109db6aa12856) with new motor.js code. It replaces the contents of ```./node_modules/johnny-five/lib/motor.js``` and then you can just use this code in your program:

````js
const {Board, Expander, Motors} = require('johnny-five');
const raspio = require('raspi-io').RaspiIO;

const board = new Board({
    io: new raspio()
});

board.on("ready", function() {

  var tb6612 = new Motors([
    { controller: "PCA9685_Hybrid", pins: { dir: 11, pwm: 4 }, invertPWM: true },
    { controller: "PCA9685_Hybrid", pins: { dir: 12, pwm: 5 }, invertPWM: true }
  ]);

  tb6612.forward(255);
  board.wait(3000, ()=> tb6612.stop);
});
````
While I don't have the same hardware to make sure it actually turns the motors, I was able to run the code and confirm it does not error out anymore. Once we get the pins worked out, I'll make a PR.

5ocjal, 2019-09-15:
There is no error, but after running the code nothing actually happened, so I thing that those pins might be not right. 

```
pi@raspberrypi:~/Luke $ sudo node motors.js 
1568562430816 Available Raspi IO  
1568562430828 Connected Raspi IO  
1568562430842 Repl Initialized  
>> 
```

dtex, 2019-09-15:
One real quick test in case we have the pins right, try moving the motors at half speed instead of full.

5ocjal, 2019-09-15:
It works!  You're amazing man! :)
The problem is only that after 3 sec it did not stop. 'Ctrl + c' didn't stop it either, I had to unplug it from power supply.

dtex, 2019-09-15:
That's awesome!!!

Remove ```, invertPWM: true``` from both motor opts and it should stop.

5ocjal, 2019-09-15:
Unfortunately, it didn't work, it's still in infinite run mode. :)

dtex, 2019-09-15:
Okay, let me think about what this could be. It might take a while so don't wait up.

5ocjal, 2019-09-15:
Okay, thanks for the help, it would take me months to solve this problem.

5ocjal, 2019-09-15:
Ok, I fix it with a little workaround.

```
  tb6612.forward(110);
  board.wait(3000, ()=> tb6612.forward(0));
```

dtex, 2019-09-15:
Okay, good to know. Does it go full speed with ```tb6612.forward(255)```?
Okay, good to know. Does it go in reverse with ```tb6612.reverse(255)```?

5ocjal, 2019-09-16:
Full speed forward works fine, but when I try to code "reverse(255)" the direction did not change, and it still goes forward.

dtex, 2019-09-16:
Okay, it sounds like we have the wrong MA and MB pins. Could you take high-resolution pictures of both the front and back of the hat? Maybe we can just trace the pins on the board?

5ocjal, 2019-09-16:
[image]
[image]

dtex, 2019-09-16:
Cool, try this.

````js
var tb6612 = new Motors([
    { controller: "PCA9685_Hybrid", pins: { dir: "GPIO17", pwm: 4 }, invertPWM: true },
    { controller: "PCA9685_Hybrid", pins: { dir: "GPIO27", pwm: 5 }, invertPWM: true }
  ]);
````

5ocjal, 2019-09-17:
Sorry for late reply - busy day. Strange things after that change happens. 
When i set "invertPWM = true" it's goes only backward no matter is it forward(125) or rev(125), whats more it ignores "board.wait...", and after 3 second runs in the same direction, but with a full speed.

When I set "invertPWM = false", everything work just fine except that when I set forward() it goes backward, and when I set reverse() it goes  forward. 

```
  tb6612.forward(120);
  board.wait(3000, ()=> tb6612.forward(0));
```

dtex, 2019-09-17:
> When I set "invertPWM = false", everything work just fine except that when I set forward() it goes backward, and when I set reverse() it goes forward.

That's great! This is a pretty common scenario and the solution is to just reverse the wires on the motors, but I assume these are all hardwired with plugs and stuff and there is no easy way to do that. Correct?

5ocjal, 2019-09-17:
You're right. But it's not a big deal - I just need to remember about it while coding.
Thank you once more - I wouldn't make it by my own. 

You should think about making a Raspberry and Johnny-Five course on Udemy or someting. I'll be first to buy it! :)

dtex, 2019-09-17:
@5ocjal I've made a PR to J5 to add support for this kind of motor. I wonder if you could test on hardware for me. Probably the easiest way to test is to just grab this version of motor.js

5ocjal, 2019-09-18:
Yeah, sure thing! I'll check it out this evening.

5ocjal, 2019-09-18:
Here are the tests results:

```
  var tb6612 = new Motors([
  var tb6612 = new Motors([
    { controller: "PCA9685_Hybrid", pins: { dir: "GPIO17", pwm: 4 }, invertPWM: true },
    { controller: "PCA9685_Hybrid", pins: { dir: "GPIO27", pwm: 5 }, invertPWM: true }
  ]);

  tb6612.forward(120);
  board.wait(3000, ()=> tb6612.forward(0));
});
  ]);
```

Goes backward, first with a speed of 120 value, then doesn't stop but make an infinity run with a full speed backward.

```
  var tb6612 = new Motors([
    { controller: "PCA9685_Hybrid", pins: { dir: "GPIO17", pwm: 4 }, invertPWM: **false** },
    { controller: "PCA9685_Hybrid", pins: { dir: "GPIO27", pwm: 5 }, invertPWM: **false** }
  ]);

  tb6612.forward(120);
  board.wait(3000, ()=> tb6612.forward(0));
```

Goes backward with a speed of 120 value, stops after 3 sec. Everything okey except of direction.

```
var tb6612 = new Motors([ 
  Motors.SHIELD_CONFIGS.PICAR_V.A, 
  Motors.SHIELD_CONFIGS.PICAR_V.B
]);
```

Throws an error:

```
pi@raspberrypi:~/Luke/js $ sudo node motors.js 
1568831695991 Available Raspi IO  
1568831696003 Connected Raspi IO  
1568831696017 Repl Initialized  
>> /home/pi/Luke/js/motors.js:12
    Motors.SHIELD_CONFIGS.PICAR_V.A, 
                          ^

TypeError: Cannot read property 'PICAR_V' of undefined
    at Board.<anonymous> (/home/pi/Luke/js/motors.js:12:27)
    at Board.emit (events.js:203:15)
    at Board.EventEmitter.emit (domain.js:448:20)
    at process._tickCallback (internal/process/next_tick.js:61:11)
    at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
```

dtex, 2019-09-20:
Excellent. That's what I needed. I've got my PR in.

If you're able to tell me which pins the pan, tilt, and steering servos connect to, we could add a RTR example for the whole robot.

5ocjal, 2019-09-20:
Sure! Here's the servo.js I've made:

```
const {Board, Servo} = require('johnny-five');
const raspio = require('raspi-io').RaspiIO;

const board = new Board({
    io: new raspio()
});
const controller = "PCA9685";

board.on('ready', ()=>{

    const servoCamX = new Servo({
        controller,
        pin: 0,
    },

    const servoCamY = new Servo({
        controller,
        pin: 2,
    },

    const servoWheels = new Servo({
        controller,
        pin: 1,
    },

    servoCamX.to(0);
    servoCamY.to(0);
    servoWheels.to(0);

})
```

closed 2019-10-06 by dtex

malixsys, 2019-11-02:
My servos:

```
function createFrontWheelsServo() {
    return new Servo({
        pin: 0,
        controller: 'PCA9685',
        range: [50, 170],
        startAt: 110
/*
        startAt: 64
*/
    });
}

function createPanServo() {
    return new Servo({
        pin: 1,
        controller: 'PCA9685',
        range: [20, 180],
        startAt: 110
/*
        startAt: 52.5
*/
    });
}

function createTiltServo() {
    return new Servo({
        pin: 2,
        controller: 'PCA9685',
        startAt: 90,
        range: [0, 180]
    });
}
``` 
but wheels do not turn with: ```    var motors = new Motors([
        {controller: "PCA9685_Hybrid", pins: {dir: "GPIO17", pwm: 4}, invertPWM: false},
        {controller: "PCA9685_Hybrid", pins: {dir: "GPIO27", pwm: 5}, invertPWM: false}
    ]);
``` 
just hangs 😢 

Turn fine with python...

dtex, 2019-11-02:
Can you post the Python code somewhere?

malixsys, 2019-11-02:
https://github.com/sunfounder/SunFounder_PiCar/blob/master/picar/back_wheels.py

malixsys, 2019-11-02:
Maybe the `bus`?

dtex, 2019-11-02:
The servos work?

malixsys, 2019-11-02:
Yes!

dtex, 2019-11-02:
And when you say "It hangs" does that mean it freezes up and the servos stop working or that the servos keep working but the wheels just don't turn. Also, can you share your whole JS program?

malixsys, 2019-11-03:
Can't do anything through my SSH after, no output change and CTRL-C breaks the SSH session...

```
const temporal = require("temporal");
const {Board, Motor, Motors, Servo} = require("johnny-five");
const Raspi = require("raspi-io").RaspiIO;
const board = new Board({
    io: new Raspi()
});

temporal.on("idle", function () {
    console.log("Temporal is idle");
});

function createFrontWheelsServo() {
    return new Servo({
        pin: 0,
        controller: 'PCA9685',
        range: [50, 170],
        startAt: 110
    });
}

function createPanServo() {
    return new Servo({
        pin: 1,
        controller: 'PCA9685',
        range: [20, 180],
        startAt: 110
    });
}

function createTiltServo() {
    return new Servo({
        pin: 2,
        controller: 'PCA9685',
        startAt: 90,
        range: [0, 180]
    });
}

console.warn('Waiting for board...');
board.on("ready", () => {
    console.warn('Board ready...');

    const motors = new Motors([
        {controller: "PCA9685_Hybrid", pins: {dir: "GPIO17", pwm: 4}, invertPWM: false},
        {controller: "PCA9685_Hybrid", pins: {dir: "GPIO27", pwm: 5}, invertPWM: false}
    ]);

    motors[0].on("start", () => {
        console.log(`start: ${Date.now()}`);
    });

    motors[0].on("stop", () => {
        console.log(`Stop: ${Date.now()}`);
    });

    const tilt = createTiltServo();

    const pan = createPanServo();

    const wheels = createFrontWheelsServo();

    board.repl.inject({tilt,  pan,  wheels, motors});

    motors.forward(50);

    board.on('exit', () => {
        /*
                tilt.center();
                pan.center();
                wheels.center();
                motors.stop();
        */
    });
});
```
Results in : 
```>> Stop: 1572808452391
start: 1572808452392
packet_write_wait: Connection to 192.168.0.50 port 22: Broken pipe```

dtex, 2019-11-03:
Thanks, that helps. Sorry to be asking so many questions but I'm kinda stumped so it's really all I can do. Rather than make one suggestion at a time, I'm going to do a brain dump:

1. Why is temporal there? It doesn't look like it's doing anything, but when it does have a task in the queue it will completely consume the machine's resources. 
1. Try removing everything that's not motor code (repl, console.log, console.warn, temporal and servos). Does it still behave the same?
1. There were some recent issues in raspi-io's i2c implementation. I'm not sure if they relate to this, but make sure you are up to date on everything.
1. Some of those raspi-io issues were actually pigpio issues (though I think those were specific to the pi 4) so what versions are you running? (J5, raspi-io, node pigpio, pigpio)

malixsys, 2019-11-03:
Thank for the help, actually I retried and the `picar rear-wheel-test` doesn't work anymore, make a weird brrrr sound and hangs

dtex, 2019-11-03:
Ah, could be a power issue. Everything runs off AA batteries on the Picar right?