rwaldron/johnny-five #1631: Use Johnny-five with Adafruit_TiCoServo & ConfigurableFirmata

kriztupher1, 2019-10-06:
Hello, My project involves a thermometer and servos connected to my arduino mega 2560. Every time the temperature event fires (every 5 seconds), the servo would click slightly. I finally found this article today explaining why it's happening: 

_“Arduino’s Servo library on AVR uses interrupts; if interrupts are disabled for any amount of time, the servos will twitch.
1-Wire protocol (used by DS18B20) needs to disable interrupts for the protocol to work. Therefore, servos glitch. (https://arduino.stackexchange.com/questions/49421/arduino-reading-ds18b20-interferes-with-servo/49422#49422)"_

So, to replace servo.h with Adafruit_TiCoServo, as the documentation suggests, I changed all references of "#include <Servo.h>" to "#include <Adafruit_TiCoServo.h>" and "Servo myservo;" to "Adafruit_TiCoServo myservo;" throughout the configurablefirmata code. I finally got it to compile, but now when I call it from johnny-five, the servo isn't moving. 

Are there references I need to change in johnny-five as well to get this working? I thought they were pretty decoupled, but it seems that isn't the case. Any help is appreciated! 

Thanks, 
Chris

dtex, 2019-10-06:
TiCoServo is an incomplete replacement for servo.h. Johnny-Five uses [writeMicroseconds](https://www.arduino.cc/en/Reference/ServoWriteMicroseconds) which TiCoServo does not support. 

The correct thing to do here would be to add that method to TiCoServo. It would not be particularly hard (less logic than write even), but there's a note from @ladyada in [a closed issue](https://github.com/adafruit/Adafruit_TiCoServo/issues/1#issuecomment-217341246) that says the library is offered as-is, and is not supported.

Nevertheless, I've [opened an issue](https://github.com/adafruit/Adafruit_TiCoServo/issues/3) to see if they would accept a PR.

kriztupher1, 2019-10-06:
Hey Donovan, 

Thanks for the reply and for the assistance. I’m looking at adafruit_ticoservo.h now. Would you mind telling me exactly what I need to alter in the code to make this work? I can just go in and make the change. Again, much appreciated. 

-Chris

dtex, 2019-10-12:
@kriztupher1 Finally got around to testing. I scoped this and confirmed that using TiCoServo with firmata does not set a pwm pulse. Not sure if it's a TiCoServo thing or a firmata thing. Digging in, but wanted to give you an update.

dtex, 2019-10-12:
Oh wait, it does work!

I dove into their example code and saw the following comment:

````c
// Servo parameters. These have limited configurability. Compatible pins,
// for example, are determined by the specific hardware. On Arduino Uno,
// Diecimila and Adafruit Pro Trinket, pins 9 and 10 are supported by the
// library. On Arduino Leonardo: pins 5, 9, 10 and 11. Adafruit Flora:
// pins D9, D10. PJRC Teensy 2.0 (not Teensy++ or 3.X): pins 4, 9, 14, 15.
// Arduino Mega: 2, 3, 5, 6, 7, 8, 11, 12, 13, 44, 45, 46.
````
I just switch to pin 9 (instead of 3) on my Uno and it worked. Which pin are you using on your mega?