rwaldron/johnny-five #1636: contrib-gpio Error: Could not locate the bindings file.

m-elias, 2019-10-16:
https://github.com/rwaldron/johnny-five/wiki/Getting-started-with-Raspberry-Pi-node-red-and-IoT
Following this guide on my Pi Zero W with the latest raspbian I get:
```
pi@thermopi:~ $ sudo node-red-start

Start Node-RED

Once Node-RED has started, point a browser at http://192.168.12.153:1880
On Pi Node-RED works better with the Firefox or Chrome browser

Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot

To find more nodes and example flows - go to http://flows.nodered.org
16 Oct 15:53:01 - [info]

Welcome to Node-RED
===================

16 Oct 15:53:01 - [info] Node-RED version: v1.0.1
16 Oct 15:53:01 - [info] Node.js  version: v10.15.2
16 Oct 15:53:01 - [info] Linux 4.19.75+ arm LE
16 Oct 15:53:24 - [info] Loading palette nodes
16 Oct 15:53:48 - [warn] ------------------------------------------------------
16 Oct 15:53:48 - [warn] [node-red-contrib-gpio/gpio] Error: Could not locate the bindings file. Tried:
 → /root/.node-red/node_modules/serialport/build/serialport.node
 → /root/.node-red/node_modules/serialport/build/Debug/serialport.node
 → /root/.node-red/node_modules/serialport/build/Release/serialport.node
 → /root/.node-red/node_modules/serialport/out/Debug/serialport.node
 → /root/.node-red/node_modules/serialport/Debug/serialport.node
 → /root/.node-red/node_modules/serialport/out/Release/serialport.node
 → /root/.node-red/node_modules/serialport/Release/serialport.node
 → /root/.node-red/node_modules/serialport/build/default/serialport.node
 → /root/.node-red/node_modules/serialport/compiled/10.15.2/linux/arm/serialport.node
16 Oct 15:53:48 - [warn] ------------------------------------------------------
16 Oct 15:53:48 - [info] Settings file  : /root/.node-red/settings.js
16 Oct 15:53:48 - [info] Context store  : 'default' [module=memory]
16 Oct 15:53:48 - [info] User directory : /root/.node-red
16 Oct 15:53:48 - [warn] Projects disabled : editorTheme.projects.enabled=false
16 Oct 15:53:48 - [info] Flows file     : /root/.node-red/flows_thermopi.json
16 Oct 15:53:48 - [info] Creating new flow file
16 Oct 15:53:48 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

16 Oct 15:53:48 - [info] Starting flows
16 Oct 15:53:48 - [info] Started flows
16 Oct 15:53:49 - [info] Server now running at http://127.0.0.1:1880/
```

m-elias, 2019-10-16:
node-Arduino was working except I was looking for OneWire support.

dtex, 2019-10-16:
This is a serialport issue. Perhaps [this post](https://github.com/serialport/node-serialport/issues/1271#issuecomment-321572626) will help?

m-elias, 2019-10-16:
I'll check it out when I try contrib-gpio again.
If I install contrib-johnny-five instead of contrib-gpio (they seem to be used interchangeably on some blogs) then my node-red startup is good but my j5-platform node has a ""path" is not defined" error.

closed 2020-10-24 by stale[bot]