rwaldron/johnny-five #1642: Handling wrong ports when initializing boards.

razor9876, 2019-11-12:
Hi, so my short question is: 
Is it possible to add new board to `five.Boards()` in run? like `boards.push(new five.Board())`?

Long explanation is that I'm trying to control my boards via website that I wrote, the problem is that if I send wrong port to my backend, listener(`boards.on('error')`) will return error - and that's okay. But then if I want to add new board with correct port I will get error anyway since wrong port is already initialized in `boards`. There is so much variances that my boards will stop responding(comunicating), it would be confusing if I describe all of them. But it's okay to assume that if wrong port will be initialized, whole application is getting trouble with comunication with boards. Only way to get my application working correctly is to initialize all boards without any mistakes in ports.

Is there a way to disconnect, close already initialized board?
Is there a way to free port after initializing port? (creating new board with port "COM3" for example, and after - disconnecting it)

Or there is no walkaround and you just HAVE to initialize all boards with correct ports with first try?

dtex, 2019-11-12:
You can't disconnect and reconnect to a board, but I think it is possible to find the right port before you initialize it in the first place. I assume we are talking about Arduinos running firmata connected via serial right?

If so, I think you can rename the firmata sketch for each board (instead of firmata.ino call it like boardone.ino, boardtwo.ino). I then think you can use some combination of serialport.list() and firmata.js to identify each board by its sketch name.

I don't have any hardware with me here at work (which I need to get back to), but I can try it out tonight.

It may be that you connect, determine which board it is, and then start instantiating the devices on the right boards... I can't remember, but it's come up before.

dtex, 2019-11-12:
Also there was this trick:

https://github.com/rwaldron/johnny-five/issues/883#issuecomment-362413728

razor9876, 2019-11-12:
Well, but what about ESP wifi module. I have to send (POST from my website) IP and port to my server, but if I make a mistake everyting will fall apart and comunication with other boards will be no longer possible. And then only possibility to get things working again is restarting my server.

And I'm not going to lie, but I have no idea what firmata sketch is.

closed 2020-11-22 by stale[bot]