rwaldron/johnny-five #1643: Support for serialport version 8

eMarek, 2019-11-30:
I am curious if there are any plans to upgrade `johnny-five` with dependancy on `serialport` version 8 and higher. I am especially interested in version 8.0.4 which now supports node 12.  Reason why I am asking is because `electron-rebuild` fails at building `serialport` version ^7.1.5 which `johnny-five` currently uses. More about it here: https://github.com/electron/electron-rebuild/issues/320

dtex, 2019-12-01:
We should solicit input from @rwaldron and @reconbot on this

eMarek, 2019-12-01:
@rwaldron, @reconbot, pretty please. :)

reconbot, 2019-12-01:
It can totally work, the breaking changes are not really relevant for J5 but I’d want to step through the upgrade guide

eMarek, 2019-12-02:
I don't know if it helps you anyhow, but when I tried to run J5 with latest serialport (8.0.5) I saw some errors related to `SerialPort.list()`, but I didn't have time to debug further.

reconbot, 2019-12-03:
it now returns a promise instead of taking a callback

AlexandreMouyen, 2019-12-08:
> I don't know if it helps you anyhow, but when I tried to run J5 with latest serialport (8.0.5) I saw some errors related to `SerialPort.list()`, but I didn't have time to debug further.

I try and hot fix it and it seems to work. This problem took all my day...

reconbot, 2019-12-08:
Glad you figured it out. Seriaport gives an error explaining the change.

eMarek, 2019-12-08:
@AlexandreMouyen, do you mind sharing your fix? Did you dig into johnny five and fixed it here or have you found a work around?

reconbot, 2019-12-09:
I believe we need to update firmata and johnny five for this. 

This is an untested PR for fimrata https://github.com/firmata/firmata.js/pull/201
This is an untested PR for Johnny Five https://github.com/rwaldron/johnny-five/pull/1645

AlexandreMouyen, 2019-12-09:
> @AlexandreMouyen, do you mind sharing your fix? Did you dig into johnny five and fixed it here or have you found a work around?

Yeah, I will see PR of all of this & try to fix for everybody. I just did local changes on my node_modules directory, it is looks like to work.

AlexandreMouyen, 2019-12-09:
It will need to wait, I still have a problem...

[image]

eMarek, 2019-12-10:
I am not sure if you are using electron or not, but this exact problem is solved with `electron-rebuild` which we are trying to get working with latest serialport.
https://electronjs.org/docs/tutorial/using-native-node-modules

AlexandreMouyen, 2019-12-10:
I know it thanks, and now on one of my computer it's working. Not on the other one...

eMarek, 2019-12-10:
Got it. Let's wait for pull request to be merged and new version released. Finger crossed everything will work after that.

AlexandreMouyen, 2019-12-10:
> Got it. Let's wait for pull request to be merged and new version released. Finger crossed everything will work after that.

Yeah I am waiting this too.

eMarek, 2020-01-07:
I've tested the latest `johnny-five` version `1.4.0` and it works. Thank you to @reconbot!

closed 2020-01-07 by eMarek

reconbot, 2020-01-07:
🙌👏🙌