rwaldron/johnny-five #1644: Wrong auto connect to port

AlexandreMouyen, 2019-12-07:
Hello,

My "new five.Board()" auto connect is not working correctly. The lib tries to connect to the COM1 instead of the COM3 where my Arduino Leonardo is.

[image]

Where I manually set the COM3, it is working properly.

Firmata 2.5.8.
Johnny Five 1.3.1

Do you have an idea?

Thanks.

Apoorve73, 2020-01-05:
Hello Sir! May be the baud rate isn't matching or I would prefer doing it manually.

closed 2021-07-21 by stale[bot]

vilelalabs, 2022-01-11:
> Hello,
> 
> My "new five.Board()" auto connect is not working correctly. The lib tries to connect to the COM1 instead of the COM3 where my Arduino Leonardo is.
> 
> [image]
> 
> Where I manually set the COM3, it is working properly.
> 
> Firmata 2.5.8. Johnny Five 1.3.1
> 
> Do you have an idea?
> 
> Thanks.

did you solved? getting this error too.