rwaldron/johnny-five #1648: HC-05 bluetooth connection timeout with Arduino Micro

anuragphadke, 2020-01-01:
Hi,
I was able to successfully pair HC-05 with Arduino Micro as suggested in: https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-HC-05-Bluetooth-Serial-Port-Module

I can then pair successfully to my macbook-pro and get the Bluetooth connected icon. However, running the snippet to communicate via JohnnyFive breaks the bluetooth pairing with Macbook pro and returns the error below: https://gist.github.com/anuragphadke/7361a19fdeb630243839f6aa6b5fc2e1 
```
1577906249124 Connected /dev/cu.BotMe2-SPPDev
1577906259131 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
events.js:173
    throw err; // Unhandled 'error' event
    ^

Error [ERR_UNHANDLED_ERROR]: Unhandled error. ([object Object])
    at Board.emit (events.js:171:17)
    at Board.EventEmitter.emit (domain.js:442:20)
    at Board.log (/Users/aphadke/webfast/appium-arduino-driver/node_modules/johnny-five/lib/board.js:648:8)
    at Board.(anonymous function) [as error] (/Users/aphadke/webfast/appium-arduino-driver/node_modules/johnny-five/lib/board.js:659:14)
    at Board.<anonymous> (/Users/aphadke/webfast/appium-arduino-driver/node_modules/johnny-five/lib/board.js:390:14)
    at ontimeout (timers.js:424:11)
    at tryOnTimeout (timers.js:288:5)
    at listOnTimeout (timers.js:251:5)
    at Timer.processTimers (timers.js:211:10)
```

Things I have tried so far:
- resetting via 2seconds as well as long 11second.
- uploading various Firmata after the initial AT handshake as described in Step 3 at https://github.com/rwaldron/johnny-five/wiki/Getting-Started-with-Johnny-Five-and-HC-05-Bluetooth-Serial-Port-Module (StandardFirmata, StandardFirmataPlus, as well as the Firmata mentioned here: https://github.com/rwaldron/johnny-five/issues/423#issuecomment-117311491)

Any suggestions/pointers on what I could be doing wrong here and possible fix?

anuragphadke, 2020-01-01:
consolidating few similar unresolved issues:
https://stackoverflow.com/questions/40923908/johnny-five-over-bluetooth-connection-issue
https://github.com/rwaldron/johnny-five/issues/1257

anuragphadke, 2020-01-18:
bump: any update on this issue?

dtex, 2020-01-18:
I believe @monteslu is the most knowledgeable re: formats over bluetooth, but I also know he is speaking at a conference this week and might not be able to offer any tips right away.

rwaldron, 2020-01-22:
I have no experience with that module :|

closed 2021-07-21 by stale[bot]