rwaldron/johnny-five #1649: Feather HUZZAH (ESP8266) servos not working

cburn, 2020-01-03:
Hi, 
I’m using a Huzzah board flashed with StandardFirmataWifi but can’t get a servo to work. 
I can get the servo working using the standard Arduino servo example, so I know the servo works and my working is correct, but the same wiring does not work with J5.
I can get leds to blink and a photoresister sensor to work through J5 and StandardFirmataWifi, so I know the connection is working. But when I try to sweep the servo, it doersn’t move, or give any error. 
Any ideas? I can’t find examples of this working anywhere, the closest I’ve read is the Huzzah pwm resolution is 10bit not 8, could this be the problem?

Thanks in advance,
Chris

dtex, 2020-01-03:
Hi Chris. This doesn’t sound like the 8/10 bit issue. are you able to position the servo one time? eg servo.to(180);

cburn, 2020-01-03:
No, it doesn't move on servo.to() commands either.

cburn, 2020-01-07:
After a bit of debugging, it looks like the ESP8266 servo lib doesn't define MAX_SERVOS. 
Servo was fully working after I defined MAX_SERVOS.

This issue can be closed, and I'll open one in the https://github.com/esp8266/Arduino project.

https://github.com/esp8266/Arduino/blob/master/libraries/Servo/src/Servo.h

Thanks.

closed 2021-07-21 by stale[bot]