rwaldron/johnny-five #1651: lsm303d imu compatibility

patonz, 2020-01-08:
Hi all,
ive found this [EBAY MODULE IMU 9DOF](https://www.ebay.it/itm/Module-IMU-9DOF-L3GD20-LSM303D-gyroscope-accelerometre-magnetometre-I2C-arduino/323634145156?hash=item4b5a192784:m:mjbfL5E_XGN3OOpBBF8yCiw) and i recognize the LSM303D actually its not supported by default from the lib. 
Any one have used this with johnny five? there are some difference on registers or can simply instanciated with the old one 'lsm303c'?

dtex, 2020-01-08:
There is some overlap between the register maps, but I can't tell if it will just work. I'm pretty sure you'd be missing out on some of the LSM303D features.

patonz, 2020-01-08:
> There is some overlap between the register maps, but I can't tell if it will just work. I'm pretty sure you'd be missing out on some of the LSM303D features.

Ok then i will try and post the result^^, thx

rwaldron, 2020-01-09:
I've ordered two different LSM303D breakouts to help with support.

rwaldron, 2020-01-10:
🤦‍♂ I just found 2 of these that I already had

patonz, 2020-01-10:
> 🤦‍♂ I just found 2 of these that I already had

"shit happens" cit. :P

Workshop2, 2020-05-21:
Hey, I don't suppose anyone got this working?

I have a [Adadruit LSM303](https://www.adafruit.com/product/1120) and I can't seem to get anything to work. It was the recommended chip in this API doc: http://johnny-five.io/api/imu/

closed 2022-01-09 by stale[bot]