rwaldron/johnny-five #1656: > Greetings, why do not I see the original keypad of arduino? to program it, I'm talking about the one they always use in YouTube tutorials, I think it's called keypad 4 x 4 Matrix Array 16 Key Membrane?

AlexKutschera, 2020-02-06:
> Greetings, why do not I see the original keypad of arduino? to program it, I'm talking about the one they always use in YouTube tutorials, I think it's called keypad 4 x 4 Matrix Array 16 Key Membrane?

Agree. The 4x4 is very popular and cheap. I am surprised it is not supported - or if it is, I don't know how to configure it.
If somebody can help, here are the specs:
https://www.parallax.com/sites/default/files/downloads/27899-4x4-Matrix-Membrane-Keypad-v1.2.pdf
And if we need to write the controller, I can help.

_Originally posted by @davidxll in https://github.com/rwaldron/johnny-five/issues/1492#issuecomment-436500554_

AlexKutschera, 2020-02-06:
Was the required controller added and if so how's it called?

dtex, 2020-02-09:
@AlexKutschera It looks like that keypad is handled using an i2c backpack

http://johnny-five.io/api/keypad/#4x4-keypad

closed 2022-01-09 by stale[bot]