rwaldron/johnny-five #1657: BME280 initial initialization failure, nodejs, altimeter

ksteinberg, 2020-02-09:
New BME280 installed in i2C bus on Raspberry pi 3+.  Throwing an init error in the i2cWrite call of the j5-io/index.js file.

**Nodejs Code:**
  dischargeRH = new five.Multi({
      controller: "BME280"
   });

**Error:**
/home/pi/Public/agro/node_modules/j5-io/dist/index.js:480
            throw new Error('Invalid arguments');
            ^

Error: Invalid arguments
    at J5IO.i2cWrite (/home/pi/Public/agro/node_modules/j5-io/dist/index.js:480:19)
    at EventEmitter.value (/home/pi/Public/agro/node_modules/johnny-five/lib/imu.js:1554:12)
    at Object.Drivers.get (/home/pi/Public/agro/node_modules/johnny-five/lib/imu.js:2464:12)
    at Altimeter.value (/home/pi/Public/agro/node_modules/johnny-five/lib/altimeter.js:106:36)
    at new Altimeter (/home/pi/Public/agro/node_modules/johnny-five/lib/altimeter.js:189:10)
    at /home/pi/Public/agro/node_modules/johnny-five/lib/imu.js:33:24
    at Array.forEach (<anonymous>)
    at IMU.Components (/home/pi/Public/agro/node_modules/johnny-five/lib/imu.js:29:19)
    at IMU.value (/home/pi/Public/agro/node_modules/johnny-five/lib/imu.js:2617:20)
    at new IMU (/home/pi/Public/agro/node_modules/johnny-five/lib/imu.js:2707:10)

**Hardware:**
https://www.sparkfun.com/products/13676

**Environment:**
npm version
{
  agro: '1.0.0',
  npm: '6.13.6',
  ares: '1.15.0',
  brotli: '1.0.7',
  cldr: '36.0',
  icu: '65.1',
  llhttp: '2.0.1',
  modules: '79',
  napi: '5',
  nghttp2: '1.40.0',
  node: '13.7.0',
  openssl: '1.1.1d',
  tz: '2019c',
  unicode: '12.1',
  uv: '1.34.1',
  v8: '7.9.317.25-node.28',
  zlib: '1.2.11'
}

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

 npm list
agro@1.0.0 /home/pi/Public/agro
├─┬ axios@0.19.2
│ └─┬ follow-redirects@1.5.10
│   └─┬ debug@3.1.0
│     └── ms@2.0.0 deduped
├─┬ body-parser@1.19.0
│ ├── bytes@3.1.0
│ ├── content-type@1.0.4
[... 240 lines cut ...]
│ │ ├── raspi-peripheral@3.0.4 deduped
│ │ └── serialport@8.0.7 deduped
│ └─┬ raspi-soft-pwm@6.0.2
│   ├── j5-io-types@2.5.0 deduped
│   ├── pigpio@1.3.0 deduped
│   ├── raspi-board@7.3.0 deduped
│   └── raspi-peripheral@3.0.4 deduped
└─┬ ssh2-sftp-client@5.0.2
  ├─┬ concat-stream@2.0.0
  │ ├── buffer-from@1.1.1
  │ ├── inherits@2.0.3 deduped
  │ ├─┬ readable-stream@3.5.0
  │ │ ├── inherits@2.0.3 deduped
  │ │ ├─┬ string_decoder@1.3.0
  │ │ │ └── safe-buffer@5.2.0
  │ │ └── util-deprecate@1.0.2
  │ └── typedarray@0.0.6
  ├── retry@0.12.0
  └─┬ ssh2@0.8.7
    └─┬ ssh2-streams@0.4.8
      ├─┬ asn1@0.2.4
      │ └── safer-buffer@2.1.2 deduped
      ├─┬ bcrypt-pbkdf@1.0.2
      │ └── tweetnacl@0.14.5
      └── streamsearch@0.1.2

dtex, 2020-02-09:
Okay, I think I know what happened here. It doesn't look like j5-io handles ```io.i2cWrite( byte, byte, byte )```. This is probably due to a small mistake in the firmata.js documentation (J5-IO's API is based off firmata.js) which says:

````md
- `board.i2cWrite(address, register, bytes)`
  Write a single byte to the specified register.
````

It should say

````
- `board.i2cWrite(address, register, byte)`
  Write a single byte to the specified register.
````

@nebrius Would you have some free time to add this signature? I'll go submit a PR to fix the firmata.js documentation. If you're too busy I'll take a stab at it. Congrats on the new gig by the way.

@ksteinberg If you are in a time crunch (you know, some kind of *imurgency*) I could send a temporary hack your way.

rwaldron, 2020-02-09:
There should be:

- `i2cWriteReg(address, register, byte)`
- `i2cWrite(address, bytes)`

ksteinberg, 2020-02-09:
@dtex would appreciate a fix any way you can. I don't have enough experience with this sensor to what to provide for the call

dtex, 2020-02-10:
Temporary fix for this bug:

In ./node_modules/johnny-five/lib/imu.js change line 1554 from
````js
io.i2cWrite(address, this.REGISTER.RESET, 0xB6);
````
to
````js
io.i2cWriteReg(address, this.REGISTER.RESET, 0xB6);
````
and line 1649 from
````js
io.i2cWrite(address, this.REGISTER.MEASURE_H, 0x05);
````
to
````js
io.i2cWriteReg(address, this.REGISTER.MEASURE_H, 0x05);
````
and line 1690 from
````js
io.i2cWrite(address, this.REGISTER.MEASURE_TP, 0xB7);
````
to
````js
io.i2cWriteReg(address, this.REGISTER.MEASURE_TP, 0xB7);
````

> I don't have enough experience with this sensor to what to provide for the call

Your instantiation call looks good. If you share the rest of your code I can take a look.

nebrius, 2020-02-10:
Oof, I'm swamped for the next couple days and without access to an RPi, and it'll be Wednesday evening at the earliest.

As for the signature, J5-IO supports three signatures currently (written in TypeScript):

```typescript
public i2cWrite(address: number, byte: number): void;
public i2cWrite(address: number, inBytes: number[]): void;
public i2cWrite(address: number, register: number, inBytes: number[]): void;
```

We need to add a fourth signature:

```typescript
public i2cWrite(address: number, register: number, byte: number): void;
```

which will modify the combined signature to

```typescript
public i2cWrite(address: number, registerOrInBytes: number | number[], inBytes?: number | number[]): void {
```

and then we'll need to modify the argument swizzling logic to account for the new one.

ksteinberg, 2020-02-11:
@dtex That worked although the Altimeter reading is odd (0?) Does it require calibration/initialization or are we grabbing the wrong reg?

Barometer
  pressure     :  101.537
--------------------------------------
Hygrometer
  humidity     :  27.62
--------------------------------------
Altimeter
  feet         :  0
  meters       :  0
--------------------------------------
Thermometer
  celsius      :  19.81
  fahrenheit   :  67.66
  kelvin       :  292.96
--------------------------------------
Barometer
  pressure     :  101.535
--------------------------------------
Hygrometer
  humidity     :  27.6
--------------------------------------
Altimeter
  feet         :  0
  meters       :  0
--------------------------------------

nebrius, 2020-02-14:
I just published a new version of J5-IO with the new write signature. If you run `npm update` you should get it. Let us know if this fixes it!

MintyMods, 2020-02-16:
I am also seeing incorrect readings from the `BME280` in terms of both the `Temperature` and `Altimeter`. As per the screenshot above I am getting reading of `0` for altitude _(which I don't care about for this project)_ but I am also getting values of `0` readings for the temperature as follows:-

Running the following `BME280` J5 example code _(some formatting removed)_:-
```
const I2C_BME280_SENSOR_ADDR = (0x76); 
var multi = new five.Multi({
    controller: "BME280",
    address: I2C_BME280_SENSOR_ADDR
});
multi.on("change", function() {
    console.log("Thermometer:");
    console.log("  celsius      : ", this.thermometer.celsius);
    console.log("  fahrenheit   : ", this.thermometer.fahrenheit);
    console.log("  kelvin       : ", this.thermometer.kelvin);
    console.log("  Pressure     : ", this.barometer.pressure);
    console.log("  Humidity     : ", this.hygrometer.relativeHumidity);
    console.log("Altimeter");
    console.log("  feet         : ", this.altimeter.feet);
    console.log("  meters       : ", this.altimeter.meters);
    console.log("--------------------------------------"); 
});
```
Gives the following results:-
```
[1] Thermometer:
[1]   celsius      :  0
[1]   fahrenheit   :  32
[1]   kelvin       :  273.15
[1]   Pressure     :  95.819
[1]   Humidity     :  40.763
[1] Altimeter
[1]   feet         :  0
[1]   meters       :  0
[1] --------------------------------------
[1] Thermometer:
[1]   celsius      :  0
[1]   fahrenheit   :  32
[1]   kelvin       :  273.15
[1]   Pressure     :  95.819
[1]   Humidity     :  40.74
[1] Altimeter
[1]   feet         :  0
[1]   meters       :  0
[1] --------------------------------------

```

Here is what I am getting from Arduino UI running `BME280 I2C Test.ino`
```
Found BME280 sensor! Success.
Temp: 24.50°C		Humidity: 40.89% RH		Pressure: 99752.21 Pa
Temp: 24.52°C		Humidity: 40.94% RH		Pressure: 99746.67 Pa
Temp: 24.24°C		Humidity: 41.35% RH		Pressure: 99743.71 Pa
Temp: 24.21°C		Humidity: 41.43% RH		Pressure: 99740.28 Pa
Temp: 24.18°C		Humidity: 41.16% RH		Pressure: 99740.42 Pa
Temp: 24.16°C		Humidity: 41.08% RH		Pressure: 99750.18 Pa
Temp: 24.14°C		Humidity: 40.98% RH		Pressure: 99762.54 Pa
```

Here is what I am getting from Arduino UI running `BMx280_I2C.ino`
```
Pressure: 99765.00
Pressure (64 bit): 0.00
Temperature: 23.94
Humidity: 41.57
Pressure: 99763.00
Pressure (64 bit): 0.00
Temperature: 23.94
Humidity: 41.87
```

To be honest, I just **need** a sensor that can accurately(ish) read both Temperature **and** Humidity but seem to be really struggling with this :) 

- `TH02`  (with #1660) now outputs sensible readings for Temperature but has high (_+20_) readings for Humidity via Johnny-Five.
- `BME280` outputs sensible readings for Humidity but nothing for temperature from Johnny-Five. 

Here are the reading I am now getting from the `TH02` via J5:-

```
[1] --------------------------------------
[1] TH02 Thermometer
[1]   celsius           :  24.4
[1]   fahrenheit        :  75.92
[1]   kelvin            :  297.55
[1] --------------------------------------
[1] Hygrometer
[1]   relative humidity :  61.94
[1] --------------------------------------
[1] TH02 Thermometer
[1]   celsius           :  24.5
[1]   fahrenheit        :  76.1
[1]   kelvin            :  297.65
[1] --------------------------------------
[1] Hygrometer
[1]   relative humidity :  62
[1] --------------------------------------
```

 I refuse to give up on Johnny-five so any help with this is much appreciated.

_What is strange is that I swear at one point I thought I saw valid temperature readings from J5 for the BME280 but closed the console by accident and have not been able to reproduce it ~!_

Appreciate any push in the right direction in terms of investigating this myself further but I have checked the BME280 code against the datasheet and it looks alright to me _(maybe there is some javascript `casting` wizardry going on)_

P.S. Thank you @nebrius

nebrius, 2020-02-16:
@MintyMods can you post the output of `npm ls`?

MintyMods, 2020-02-16:
@nebrius Sure, output from npm ls is as follows:- 
<details>
  <summary>$ npm ls</summary>

> ```
> johnny-five@1.4.0 H:\Development\johnny-five
> +-- async@2.6.1
> | `-- lodash@4.17.15
> +-- UNMET OPTIONAL DEPENDENCY browser-serialport@latest
> +-- chalk@2.1.0
> | +-- ansi-styles@3.2.0
> | | `-- color-convert@1.9.0
> | |   `-- color-name@1.1.3
> | +-- escape-string-regexp@1.0.5
> | `-- supports-color@4.4.0
> |   `-- has-flag@2.0.0
> +-- color-convert@1.2.2
> +-- common-tags@1.8.0
> +-- copy-paste@1.3.0
> | +-- iconv-lite@0.4.18
> | `-- sync-exec@0.6.2
> +-- coveralls@3.0.2
> | +-- growl@1.10.5
> | +-- js-yaml@3.13.1
> | | +-- argparse@1.0.9
> | | | `-- sprintf-js@1.0.3 deduped
> | | `-- esprima@4.0.1
> | +-- lcov-parse@0.0.10
> | +-- log-driver@1.2.7
> | +-- minimist@1.2.0
> | `-- request@2.87.0
> |   +-- aws-sign2@0.7.0
> |   +-- aws4@1.7.0
> |   +-- caseless@0.12.0
> |   +-- combined-stream@1.0.6
> |   | `-- delayed-stream@1.0.0
> |   +-- extend@3.0.2
> |   +-- forever-agent@0.6.1
> |   +-- form-data@2.3.2
> |   | +-- asynckit@0.4.0
> |   | +-- combined-stream@1.0.6 deduped
> |   | `-- mime-types@2.1.15
> |   |   `-- mime-db@1.27.0
> |   +-- har-validator@5.0.3
> |   | +-- ajv@5.5.2
> |   | | +-- co@4.6.0
> |   | | +-- fast-deep-equal@1.1.0
> |   | | +-- fast-json-stable-stringify@2.0.0
> |   | | `-- json-schema-traverse@0.3.1
> |   | `-- har-schema@2.0.0
> |   +-- http-signature@1.2.0
> |   | +-- assert-plus@1.0.0
> |   | +-- jsprim@1.4.1
> |   | | +-- assert-plus@1.0.0 deduped
> |   | | +-- extsprintf@1.3.0
> |   | | +-- json-schema@0.2.3
> |   | | `-- verror@1.10.0
> |   | |   +-- assert-plus@1.0.0 deduped
> |   | |   +-- core-util-is@1.0.2 deduped
> |   | |   `-- extsprintf@1.3.0 deduped
> |   | `-- sshpk@1.14.2
> |   |   +-- asn1@0.2.3
> |   |   +-- assert-plus@1.0.0 deduped
> |   |   +-- bcrypt-pbkdf@1.0.2
> |   |   | `-- tweetnacl@0.14.5 deduped
> |   |   +-- dashdash@1.14.1
> |   |   | `-- assert-plus@1.0.0 deduped
> |   |   +-- ecc-jsbn@0.1.1
> |   |   | `-- jsbn@0.1.1 deduped
> |   |   +-- getpass@0.1.7
[... 909 lines cut ...]
> | |   |     +-- string_decoder@1.3.0
> | |   |     | `-- safe-buffer@5.2.0
> | |   |     `-- util-deprecate@1.0.2 deduped
> | |   +-- tunnel-agent@0.6.0 deduped
> | |   `-- which-pm-runs@1.0.0
> | +-- @serialport/parser-byte-length@8.0.4
> | +-- @serialport/parser-cctalk@8.0.4
> | +-- @serialport/parser-delimiter@8.0.4
> | +-- @serialport/parser-readline@8.0.4
> | | `-- @serialport/parser-delimiter@8.0.4 deduped
> | +-- @serialport/parser-ready@8.0.4
> | +-- @serialport/parser-regex@8.0.4
> | +-- @serialport/stream@8.0.4
> | | +-- @serialport/binding-mock@8.0.4 deduped
> | | `-- debug@4.1.1 deduped
> | `-- debug@4.1.1
> |   `-- ms@2.1.2
> +-- sinon@1.10.3
> | +-- formatio@1.0.2
> | | `-- samsam@1.1.3
> | `-- util@0.10.3
> |   `-- inherits@2.0.1
> `-- temporal@0.7.1
> ```
</details>

MintyMods, 2020-02-17:
Sorry, previous detail from Johnny-five project, new details from my actual project.

<details>
<summary>npm ls from correct location</summary>

> minty-hydro@1.0.0 H:\Development\MintyHydro
> +-- babel-core@6.26.3
> | +-- babel-code-frame@6.26.0
> | | +-- chalk@1.1.3
> | | | +-- ansi-styles@2.2.1
> | | | +-- escape-string-regexp@1.0.5 deduped
> | | | +-- has-ansi@2.0.0
> | | | | `-- ansi-regex@2.1.1
> | | | +-- strip-ansi@3.0.1
> | | | | `-- ansi-regex@2.1.1 deduped
> | | | `-- supports-color@2.0.0
> | | +-- esutils@2.0.3
> | | `-- js-tokens@3.0.2
> | +-- babel-generator@6.26.1
> | | +-- babel-messages@6.23.0 deduped
> | | +-- babel-runtime@6.26.0 deduped
> | | +-- babel-types@6.26.0 deduped
> | | +-- detect-indent@4.0.0
> | | | `-- repeating@2.0.1
> | | |   `-- is-finite@1.0.2
> | | |     `-- number-is-nan@1.0.1 deduped
> | | +-- jsesc@1.3.0
> | | +-- lodash@4.17.15 deduped
> | | +-- source-map@0.5.7 deduped
> | | `-- trim-right@1.0.1
> | +-- babel-helpers@6.24.1
> | | +-- babel-runtime@6.26.0 deduped
> | | `-- babel-template@6.26.0 deduped
> | +-- babel-messages@6.23.0
> | | `-- babel-runtime@6.26.0 deduped
> | +-- babel-register@6.26.0
> | | +-- babel-core@6.26.3 deduped
> | | +-- babel-runtime@6.26.0 deduped
> | | +-- core-js@2.6.11
> | | +-- home-or-tmp@2.0.0
> | | | +-- os-homedir@1.0.2
> | | | `-- os-tmpdir@1.0.2
> | | +-- lodash@4.17.15 deduped
> | | +-- mkdirp@0.5.1 deduped
> | | `-- source-map-support@0.4.18
> | |   `-- source-map@0.5.7 deduped
> | +-- babel-runtime@6.26.0
> | | +-- core-js@2.6.11 deduped
> | | `-- regenerator-runtime@0.11.1
> | +-- babel-template@6.26.0
> | | +-- babel-runtime@6.26.0 deduped
> | | +-- babel-traverse@6.26.0 deduped
> | | +-- babel-types@6.26.0 deduped
> | | +-- babylon@6.18.0 deduped
> | | `-- lodash@4.17.15 deduped
> | +-- babel-traverse@6.26.0
> | | +-- babel-code-frame@6.26.0 deduped
> | | +-- babel-messages@6.23.0 deduped
> | | +-- babel-runtime@6.26.0 deduped
> | | +-- babel-types@6.26.0 deduped
> | | +-- babylon@6.18.0 deduped
> | | +-- debug@2.6.9 deduped
> | | +-- globals@9.18.0
> | | +-- invariant@2.2.4 deduped
> | | `-- lodash@4.17.15 deduped
> | +-- babel-types@6.26.0
> | | +-- babel-runtime@6.26.0 deduped
> | | +-- esutils@2.0.3 deduped
> | | +-- lodash@4.17.15 deduped
> | | `-- to-fast-properties@1.0.3
[... 1311 lines cut ...]
> |   +-- which-module@1.0.0
> |   +-- y18n@3.2.1
> |   `-- yargs-parser@4.2.1
> |     `-- camelcase@3.0.0
> `-- webpack-dev-middleware@1.12.2
>   +-- memory-fs@0.4.1 deduped
>   +-- mime@1.6.0
>   +-- path-is-absolute@1.0.1 deduped
>   +-- range-parser@1.2.1 deduped
>   `-- time-stamp@2.2.0
> 
> npm ERR! missing: serialport@8.0.6, required by minty-hydro@1.0.0
> npm ERR! missing: serialport@8.0.6, required by firmata@2.2.0
> npm ERR! missing: serialport@8.0.6, required by johnny-five@1.4.0
> npm ERR! missing: @serialport/bindings@8.0.6, required by serialport@8.0.6
> npm ERR! missing: debug@4.1.1, required by serialport@8.0.6
> npm ERR! missing: debug@4.1.1, required by @serialport/bindings@8.0.6
> npm ERR! missing: ms@2.1.2, required by debug@4.1.1
> npm ERR! missing: node-abi@2.13.0, required by prebuild-install@5.3.3
> npm ERR! missing: chownr@1.1.3, required by tar-fs@2.0.0
> npm ERR! missing: readable-stream@3.5.0, required by tar-stream@2.1.0
> npm ERR! missing: readable-stream@3.5.0, required by bl@3.0.0
> npm ERR! missing: ms@2.1.2, required by debug@4.1.1

</details>

closed 2022-01-09 by stale[bot]