rwaldron/johnny-five #1659: Grove Motor Board V1 type `TB6612FNG` not working with GROVE_I2C_MOTOR_DRIVER

MintyMods, 2020-02-16:
Grove Motor Board version one is not working against J5 built in support for `GROVE_I2C_MOTOR_DRIVER`

[image]

I managed to get it working within Arduino using the following:-

```
#include "Grove_Motor_Driver_TB6612FNG.h"
#include <Wire.h>

MotorDriver motor;

void setup()
{
    // join I2C bus (I2Cdev library doesn't do this automatically)
    Wire.begin();
    Serial.begin(9600);
    motor.init();
}

void loop()
{
    // drive 2 dc motors at speed=255, clockwise
    Serial.println("run at speed=255");
    motor.dcMotorRun(MOTOR_CHA, 255);
    motor.dcMotorRun(MOTOR_CHB, 255);
    delay(1000);
    
    // brake
    Serial.println("brake");
    motor.dcMotorBrake(MOTOR_CHA);
    motor.dcMotorBrake(MOTOR_CHB);
    delay(1000);

    // drive 2 dc motors at speed=200, anticlockwise
    Serial.println("run at speed=-200");
    motor.dcMotorRun(MOTOR_CHA, -200);
    motor.dcMotorRun(MOTOR_CHB, -200);
    delay(1000);

    // stop 2 motors
    Serial.println("stop");
    motor.dcMotorStop(MOTOR_CHA);
    motor.dcMotorStop(MOTOR_CHB);
    delay(1000);
}
```

I did try modifying the `motor.js` based on [the code from Seed Studio](https://github.com/Seeed-Studio/Grove_Motor_Driver_TB6612FNG) but kind of got lost in the API for J5 in terms of how pins are being used with I2C boards. The `GROVE_I2C_MOTOR_DRIVER` seems to be implemented using `setPWM `and `setPin `to determine speed and direction although the API from Seed Studio shows a run method that sets both speed and direction within the same I2C write while specifying which motor channel to work with. 

```
 void MotorDriver::dcMotorRun(motor_channel_type_t chl, int16_t speed) {
    if (speed > 255) {
        speed = 255;
    } else if (speed < -255) {
        speed = -255;
    }

    if (speed >= 0) {
        _buffer[0] = GROVE_MOTOR_DRIVER_I2C_CMD_CW;
    } else {
        _buffer[0] = GROVE_MOTOR_DRIVER_I2C_CMD_CCW;
    }

    _buffer[1] = chl;
    if (speed >= 0) {
        _buffer[2] = speed;
    } else {
        _buffer[2] = (uint8_t)(-speed);
    }

    I2Cdev::writeBytes(_addr, _buffer[0], 2, _buffer + 1);
    delay(1);
}
```

All good fun :)

_Originally posted by @MintyMods in https://github.com/rwaldron/johnny-five/issues/1658#issuecomment-586633172_

MintyMods, 2020-02-16:
Confirmed as not compatible due to using different register locations.

MintyMods, 2020-02-16:
For reference the register values the board requires:-

```
    REGISTER: {
      value: {
        ADDRESS: 0x14,
      }
    },
    COMMANDS: {
      value: {
        BRAKE: 0x00,
        STOP: 0x01,
        CW: 0x02,
        CCW: 0x03,
        STANDBY: 0x04,
        NOT_STANDBY: 0x05,
        STEPPER_RUN: 0x06,
        STEPPER_STOP: 0x07,
        STEPPER_KEEP_RUN: 0x08,
        SET_ADDR: 0x11,
      },
```

MintyMods, 2020-02-16:
Picture showing the front of the [board](https://www.seeedstudio.com/Grove-I2C-Motor-Driver-TB6612FNG-p-3220.html):-

[image]

MintyMods, 2020-02-16:
Controlling the board directly using I2C does work using the following test code

```
    // "Grove_Motor_Driver_TB6612FNG.h"
    const I2C_GROVE_MOTORBOARD_ADDR = (0x14); 
    const COMMAND = {
      BRAKE: 0x00,
      STOP: 0x01,
      CW: 0x02,
      CCW: 0x03
    }
    const MOTOR = {
      A: 0,
      B: 1
    }
    let speed = 127;

    // FORWARD Motor A @ speed
    board.io.i2cWrite(I2C_GROVE_MOTORBOARD_ADDR, COMMAND.CW, [MOTOR.A, speed]);
    board.wait(1000, function() {
      
      // REVERSE Motor A
      board.io.i2cWrite(I2C_GROVE_MOTORBOARD_ADDR, COMMAND.CCW, [MOTOR.A, speed]);
      board.wait(1000, function() {
          
          // STOP Motor A
          board.io.i2cWriteReg(I2C_GROVE_MOTORBOARD_ADDR, COMMAND.STOP, MOTOR.A);
          board.wait(1000, function() {

            // REVERSE Motor B
            board.io.i2cWrite(I2C_GROVE_MOTORBOARD_ADDR, COMMAND.CCW, [MOTOR.B, (speed + 100)]);
            board.wait(1000, function() {

                // FORWARD Motor B
                board.io.i2cWrite(I2C_GROVE_MOTORBOARD_ADDR, COMMAND.CW, [MOTOR.B, (speed + 100)]);
                board.wait(1000, function() {

                    // BRAKE Motor B
                    board.io.i2cWriteReg(I2C_GROVE_MOTORBOARD_ADDR, COMMAND.BRAKE, MOTOR.B);       

                });
            });
        });
      });
    });
```

MintyMods, 2020-02-16:
If someone could point me in the right direction for what is required to provide a new `Motor Board` _(for this [Grove V1 TB6612FNG](https://www.seeedstudio.com/Grove-I2C-Motor-Driver-TB6612FNG-p-3220.html))_ as well as new `Shield` implementation for the [Atlas Tenticle Shield](https://www.whiteboxes.ch/docs/tentacle/t1/#/) within the Johnny-Five API?

I would like to contribute both if this is possible? 

They are both controlled via I2C so I just need to understand how to integrate these I2C calls into Johnny-Five callbacks? Also my current usage for the Atlas Shield is by requesting readings so not sure how this fits with the J5 event driven aspect. 

Any tips on the testing framework (grunt, spy) you guys are using are using would also be appreciated as I tried to 'un-comment' the `onchange` test within the `TH02 imu.js test` but got lost with the spy.calls being null. 

Thank you for any pointers.

dtex, 2020-02-16:
I can help a lot more with motor.js than I could with imu.js.

## Grove

The GROVE_I2C_MOTOR_DRIVER controller was built for the Grove L298 based motor controller, so we will need to distinguish the two. I would rename ```GROVE_I2C_MOTOR_DRIVER``` to ```GROVE_I2C_L298``` and then add an undocumented alias for ```GROVE_I2C_MOTOR_DRIVER``` so legacy code continues to work.

Then I would create a new controller called ```GROVE_I2C_TB6612FNG```. This new controller is unlike any motor controller that we support (and we support a lot) so you're gonna have to override some stuff, but fortunately, this class was designed with that in mind.

It's tricky because a single register is used to set the direction for both motors and we have always managed motors independently. Further, the pattern of bits used to represent direction is just completely bonkers. That's not a hard problem to solve, it just requires an unnecessarily messy conditional when setting direction (instead of just using bitwise operators on two bits). Note: I'm being whiny about this thing I don't like but it's really not a big deal to overcome.

You will also have to manage board state, not just motor state, since each board will always need to know direction data for both motors and you could have more than one of these boards attached. I would probably just add a state object to the controller that uses i2c addresses as keys so you can store state for multiple boards.

## Atlas Tentacle Shield

Except for Temperature, we don't have classes that these sensors would fit in. I think they might be best suited for a [Johnny-Five plug-in](http://johnny-five.io/examples/plugin/).

MintyMods, 2020-02-17:
Thanks again @dtex for your prompt assistance. 

I am still a little confused over the use of pins when it comes to these I2C devices. Is it simply a case of the pin API is just being re-used to specify a unique identifier to determine which of the (2 in this case) motors we are dealing with within the board.

Are there any existing example J5 plugins that I could take a look at?

I will have a play and see where I can get with this.

MintyMods, 2020-02-17:
I think I am understanding some of the basics but hitting this:-

    [1] Debugger attached.
    [1] H:\Development\MintyHydro\node_modules\firmata-io\lib\firmata.js:831
    [1]     this.pins[pin].value = value;
    [1]                          ^
    [1]
    [1] TypeError: Cannot set property 'value' of undefined
    [1]     at Firmata.pwmWrite (H:\Development\MintyHydro\node_modules\firmata-io\lib\firmata.js:831:26)
    [1]     at Motor.setPWM (H:\Development\MintyHydro\node_modules\johnny-five\lib\motor.js:795:11)
    [1]     at Motor.speed (H:\Development\MintyHydro\node_modules\johnny-five\lib\motor.js:834:10)
    [1]     at Motor.stop (H:\Development\MintyHydro\node_modules\johnny-five\lib\motor.js:867:8)
    [1]     at Motor.value (H:\Development\MintyHydro\node_modules\johnny-five\lib\motor.js:476:14)
    [1]     at new Motor (H:\Development\MintyHydro\node_modules\johnny-five\lib\motor.js:774:8)

While I am still getting to grips with this *(I seriously didn't know you could attach the Chrome debugger to node until just now!)* Stepping via my new found **debugger** friend shows it is going into the standard Motor.setPWM 

    Motor.prototype.setPWM  =  function(pin, value)  {
    this.io.analogWrite(pin,  __.map(value,  0,  255,  0,  this.board.RESOLUTION.PWM));
    };

rather than what I hoped would be my overridden function (based on nothing other than the other one works):- 

>     GROVE_I2C_TB6612FNG: {
>         REGISTER: {
>           value: {
>             ADDRESS: 0x14,
>           }
>         },
>         COMMANDS: {
>           value: {
>             SET_SPEED: 0x82,
>             SET_PWM_FREQUENCY: 0x84,
>             SET_DIRECTION: 0xAA,
>             NOOP: 0x01,
>           },
>         },
>         initialize: {
>           value: function(opts) {
>      >>>>>>> This code is executed  <<<<
>             var state = priv.get(this);
>             var shared = priv.get("GROVE_I2C_TB6612FNG");
>             if (!shared) {
>               shared = {
>                 direction: {
>                   A: 0x01,
>                   B: 0x01,
>                 },
>                 speed: {
>                   A: 0,
>                   B: 0,
>                 }
>               };
>               priv.set("GROVE_I2C_TB6612FNG", shared);
>             }
>             state.shared = shared;
>             state.pin = opts.pin.toUpperCase();
>             this.opts.pins = {
>               pwm: opts.pin,
>               dir: opts.pin,
>             };
>             this.address = opts.address || this.REGISTER.ADDRESS;
>             opts.address = this.address;
>             this.io.i2cConfig(opts);
>           },
>           setPWM: {
>             value: function(pin, value) {
>               debugger
>     <<<<< This breakpoint is not hit before the exception >>>>>
I am assuming the error is caused because the standard motor code is running rather than what should be my overridden I2C alternative? 

Is there some registration required for new boards such as the following which I remember from the `imu.js` I was recently playing with:-

    TH02:  {
    initialize:  {
    value:  function(opts)  {
    Components.call(this,  "TH02",  opts);
    }
    },
    components:  {
    value: [HYGROMETER, THERMOMETER]
    },
    },

MintyMods, 2020-02-17:
> ## Atlas Tentacle Shield
> Except for Temperature, we don't have classes that these sensors would fit in. I think they might be best suited for a [Johnny-Five plug-in](http://johnny-five.io/examples/plugin/).

I assume this would be a standalone project hosted on GitHub with NPM support for the Johnny-Five API?

MintyMods, 2020-02-19:
> I think I am understanding some of the basics but hitting this:-
> 
Please ignore my previous post regarding the breakpoints etc. 

A syntax error typo put the `setPWM` and `setPin` functions **within** the `initialize` function and I just couldn't spot the mistake:-

[image]

Correcting that and I can now see the overrides and hit the breakpoints. 

Happily moving on...

MintyMods, 2020-02-20:
### **Wow!** 
Really Sorry, I did not mean to publish all that GitHub traffic from my Fork of Johnny-Five... 

I do struggle a little with Git still, as you can tell :)

On a plus note I _think_ I have made a progress and can now control the motors using J5.

I just need to get to grips with Grunt and produce the test and I will submit a PR for review.

I still don't understand how all the `STEPPER` functionality from the board will be used but it doesn't look like this belongs in `motor.js` anyway?

dtex, 2020-02-20:
> I did not mean to publish all that GitHub traffic 

No big deal, it's just a bunch of references. If that's the worst thing that happens you're doing really good.

You're right about stepper. It has its own class and unless you need to control steppers, I wouldn't worry about it.

MintyMods, 2020-02-21:
@dtex Can you please expand on your following comment and/or point me in **_any_** direction for more information about this?

> It's tricky because a single register is used to set the direction for both motors and we have always managed motors independently. Further, the pattern of bits used to represent direction is just [...]

Currently I am just overriding `setPWM` and `setPin` and this seems to be working fine in terms of actually controlling the motor direction and speed :-

```
    setPWM: {
      value: function(pin, value) {
        var state = priv.get(this);
        var direction = state.shared.direction[state.pin];
        var speed = Board.constrain(value, 0, 255) | 0;
        state.shared.speed[state.pin] = speed;
        this.io.i2cWrite(this.address, direction, [this.MOTOR[state.pin], speed]);
      }
    },
    setPin: {
      value: function(pin, value) {
        var state = priv.get(this);
        var speed = state.shared.speed[state.pin];
        var direction = value ? this.COMMANDS.CW : this.COMMANDS.CCW;
        state.shared.direction[state.pin] = direction;
        this.io.i2cWrite(this.address, direction, [this.MOTOR[state.pin], speed]);
      }
    },
```

But I am struggling to write sensible tests for `forwardControlBoth` and `reverseControlBoth` as using more than one motor seems to complicate things and I don't see the I2C traffic I would expect from the `this.i2cWrite.getCall(X)`. Any help with this is appreciated but I suspect it has something to do with your previous comment.

> You will also have to manage board state, not just motor state, since each board will always need to know direction data for both motors and you could have more than one of these boards attached. I [...]

I need to purchase an additional board to be able to understand, work on and test this functionality but can I ask if the existing `GROVE_I2C_MOTOR_DRIVER` supports this board state as this new controller is basically just a copy of that one.

Thank you again for helping me with **all** my issues.

closed 2022-01-09 by stale[bot]