rwaldron/johnny-five #1669: Is there any J5 library for STM32F103

pritanshupatel, 2020-03-20:
STM32F103 features 
[image]

i am building a flight controller using STM32F103, will J5 support it, if yes is there any example to which i can refer.

Also if you can suggest alternative to STM32 with same capabilities which J5 supports

Thanks
Pritanshu Patel

dtex, 2020-03-20:
The STM32 is just a Microcontroller and there's no support in J5 specific to any one microcontroller. What J5 would support would be the software running on the microcontroller in combination with the hardware that microcontroller is attached to but since you're making your own that's obviously not built into Johnny-Five already. In fact, we don't have any flight controller support. But that doesn't mean you can't do this.

Say for example the STM32 is doing the actual low-level flight control (because the feedback loop in node.js + J5 can't handle that),. You could use serial or i2c communication to talk to your microcontroller from Johnny-Five for some other part of the system (i.e. GPS & wayfinding to manage high-level flight control). J5 doesn't really care what microcontroller it's talking to over those protocols so yes it will support the STM32, but you'll need to design a protocol for communication.

pritanshupatel, 2020-03-20:
Thanks for the information :)

closed 2020-03-20 by pritanshupatel