rwaldron/johnny-five #1671: Passing an array of targets to Animation constructor is not supported

dtex, 2020-04-01:
This code should work but throws: TypeError: this.target[Animation.normalize] is not a function

````js
const { Animation, Board, Led } = require("johnny-five");
const board = new Board();

board.on("ready", () => {
  const led1 = new Led(11);
  const led2 = new Led(12);

  const animation = new Animation([led1, led2]);

  animation.enqueue({
    easing: "linear",
    duration: 500,
    cuePoints: [0, 1],
    keyFrames: [ [0, 255], [0, 255] ],
    metronomic: true
  });

  board.wait(12000, () => {
    animation.stop();
  });
});
````
When normalize is called on an array we need to call normalize on each member of the array.

aditya6752, 2020-07-12:
hey, can you please tell me that what is the bug in this?

dtex, 2020-07-12:
I'm probably oversimplifying it, but [this bit of code](https://github.com/rwaldron/johnny-five/blob/master/lib/animation.js#L397-L398) should behave differently when the target is an array. Something like:

````js
// Run through the target's normalization
if (Array.isArray(this.target) {
  this.target.forEach((device, index) => {
    keyFrameSet[index] = this.target[index][Animation.normalize](keyFrameSet[index]);
  });
} else {
  keyFrameSet = this.target[Animation.normalize](keyFrameSet);
}
````

I'm sure it's more complicated than this. Will definitely need new tests.

closed 2022-01-09 by stale[bot]

ScreamZ, 2023-10-19:
Same issue for SERVO, has to use Servos or Servo.Collection