rwaldron/johnny-five #1674: How to share the jhonny five instance and the board to other files ?

ademarCardoso, 2020-04-21:
Hello my friends, I'm setting up a project for a printer and I have several separate files, for different functions of the printer, my question is the following, it would be possible for me to make the instance of jhonny five in my server.js file and share that instance as global, for all other files to access the arduino and thus control parts of the printer accessing the same instance ?

Or would I have to have a single .js file for all printer functions ?
PS: I'm using advancedFirmata and an Uno arduino

If you don't understand the question let me know, and thanks friends!

dtex, 2020-04-21:
I understand the question because I've faced the exact same conundrum before.  I recently created my first Johnny-Five plugin and I think it's a great approach for something like this. The example code for plug-ins is pre es6, but I wrote one that was done with class syntax and it worked great. 

First create a JS file for your printer:

````js
// printer.js
const eventEmitter = require("events");

module.exports = function(five) {
  return (function() {

    class Printer extends eventEmitter {

      constructor(opts) {

        // Normalize your opts here
  
        super();
        five.Board.Component.call( this, opts = five.Board.Options(opts) );
        this.initialize(opts);
      }

      /*
       * Initialize the printer
       */
      initialize(opts) {

        // Put all your initialization code here
        const servo = new five.servo(12); 
        // etc etc.

        this.on("error", (err) => {
          console.error(err.message);
        });

        this.emit("ready");
  
      }

      /*
       * Add all your methods
       */
      print(stuffToPrint) {
        // Print your stuff
      }

    return Printer;
  }());

};
````

And then use that file from your programs entry point like this:

````js
// index.js
const five = require("johnny-five");
const Printer = require("./printer.js")(five);
// You could also use ESM syntax above

const board = new five.Board();

board.on("ready", () => {

  const printer = new Printer({
    // All your opts
  });

  printer.on("ready", () => {
    console.log("The Printer is connected and ready");
    printer.print("Hello World!");
  });

  // The J5 board instance is available for other things
  // You can instantiate any class from Johnny-Five
  // or from another plugin

});
````

ademarCardoso, 2020-04-21:
Thank you very much for the answer, I will test it and if it has good results I will let you know in a moment, thank you !

closed 2020-04-21 by ademarCardoso