rwaldron/johnny-five #1675: How to create functions that are reusable with an instance of the board ?

ademarCardoso, 2020-04-21:
Hello everyone, I'm using the jhonny-five lib to create a project for a braille printer, I'm using a stepper motor and a relay, the code below is one of the printer's functions, but I can't start the motor and neither the relay the way I set it up, I need separate functions but I don’t know how to use the Board instance that way, at the moment this code doesn’t give any error, it connects to the Arduino but doesn’t do anything else. Does anyone know how to set up functions that run the relay and stepper separately, so that I can call certain parts of the code?

// Example code
```js
const five = require("johnny-five")
const board = new five.Board()
let text = [['1','0'],['0','0'],['0','0']]
let size = text.lenght

// This Loop will be executed several times, this is just an example
for (let init = 0; init < text.length; init++) {
    mark(text[init][0], 0)
  }

function next(direction, steps = 500) { // Change the steps number
  board.on("ready", () => {
    let stepperMotor = new five.Stepper({
      type: five.Stepper.TYPE.DRIVER,
      stepsPerRev: 200,
      pins: {
        step: 3,
        dir: 2
      }
    })

    stepperMotor.rpm(300).step({ steps: steps, direction: direction }, function () {
      console.log("jump to next mark")
    });
  })
  return
}

function mark(text, direction) {
  board.on("ready", () => {
    let relay = new five.Relay(4)

    if (text[0] === '0') {
      next(direction)
    }
    if (text[0] === '1') {
      relay.open()
      relay.close()
    }
    if (text[1] === '0') {
      next(direction)
    }
    if (text[1] === '1') {
      relay.open()
      relay.close()
    }
  })
}
```

ademarCardoso, 2020-04-28:
> Hello everyone, I'm using the jhonny-five lib to create a project for a braille printer, I'm using a stepper motor and a relay, the code below is one of the printer's functions, but I can't start [...]

dtex, 2020-04-28:
In most apps the board.on("ready") listener is something you attach only once when first starting your program. I can give an example later, after work.

dtex, 2020-04-29:
Sorry, I forgot to do this last night. Note that this is just one way and I did not test with hardware of course:

````js
const five = require("johnny-five")
const board = new five.Board()
let text = [['1','0'],['0','0'],['0','0']]
let size = text.lenght

board.on("ready", () => {

  // Initialize the components (happens just one time)
  const stepperMotor = new five.Stepper({
    type: five.Stepper.TYPE.DRIVER,
    stepsPerRev: 200,
    pins: {
      step: 3,
      dir: 2
    }
  });

  const relay = new five.Relay(4)

  // Let's define our methods that use those componenets
  function next(direction, steps = 500) { // Change the steps number
    stepperMotor.rpm(300).step({ steps: steps, direction: direction }, function () {
      console.log("jump to next mark"); // Happens when motor finishes move
    });
    return; // This will happen right away
  }

  function mark(text, direction) {
    if (text[0] === '0') {
      next(direction)
    }
    if (text[0] === '1') {
      relay.open()
      relay.close()
    }
    if (text[1] === '0') {
      next(direction)
    }
    if (text[1] === '1') {
      relay.open()
      relay.close()
    }
  }

  // And here is where would start to use those methods
  for (let init = 0; init < text.length; init++) {
    mark(text[init][0], 0)
  }

});
````

dtex, 2020-04-29:
I submitted a reply but quickly realized it was flawed. I didn't forget about you, it's just a little trickier. Here's an incomplete solution:
````js
const five = require("johnny-five")
const board = new five.Board()
let text = [['1','0'],['0','0'],['0','0']]
let size = text.lenght

board.on("ready", () => {

  // Initialize the components (happens just one time)
  const stepperMotor = new five.Stepper({
    type: five.Stepper.TYPE.DRIVER,
    stepsPerRev: 200,
    pins: {
      step: 3,
      dir: 2
    }
  });

  const relay = new five.Relay(4)

  // Let's define our methods that use those componenets
  function next(direction, steps = 500) { // Change the steps number
    stepperMotor.rpm(300).step({ steps: steps, direction: direction }, function () {
      console.log("jump to next mark"); // Happens when motor finishes move
    });
    return; // This will happen right away
  }

  function mark(text, direction) {
    if (text[0] === '0') {
      next(direction)
    }
    if (text[0] === '1') {
      relay.open()
      relay.close()
    }
    if (text[1] === '0') {
      next(direction)
    }
    if (text[1] === '1') {
      relay.open()
      relay.close()
    }
  }

  // Your actual program goes here

});
````
The reason I say "incomplete" is that I didn't try and add your logic code. The flaw in my earlier post and your original code was that all the commands are sent to the Arduino right away instead of waiting for each move and then waiting for each relay cycle to execute.

ademarCardoso, 2020-05-01:
Thanks @dtex  for remembering me, at the moment almost everything is working, only that the chambers go all at once, would there be any way to avoid it, or do I add setTimeOut everywhere? And thanks for the answer above, it helped me a lot :)

dtex, 2020-05-01:
I would create an array that works as a FIFO queue for commands that will be sent to your printer. You can take a single command (next, mark, reverse) from the queue, and then call an async function that executes each command.

This sounds like a super cool project. I hope you share some videos of it in action.

ademarCardoso, 2020-05-01:
I'll try to implement this today, if everything goes well, I'll send a video of everything working, I'm building a Braille printer :)

closed 2020-06-19 by ademarCardoso