rwaldron/johnny-five #1676: LCD Grove JHD1313M1 + Special characters

Hugo54, 2020-04-26:
Hi,

Special characters seems to not working. All the print function & background colors works but nothing is displayed when I use special char like heart.

**Configuration :**
os : windows 10
nodejs : v12.13 / x64
johnny-five: "1.4.0"
LCD : Grove JHD1313M1 / P131021
Arduino : Arduino UNO

`var five = require("johnny-five");
var board = new five.Board({
    port: "COM3"
});

board.on("ready", function() {
    var lcd = new five.LCD({
        controller: "JHD1313M1"
    });
    lcd.bgColor("blue");
    lcd.useChar("heart");
    lcd.print(":heart:")

    var interval = null;
    var index = 0;
});
`

Regards,

Hugo

dtex, 2020-04-26:
First, I want to make sure that you are writing:

````js
// this
lcd.print(":heart:")

// not this
lcd.print("❤️")
````

It may just be a markdown thing. I'm not sure.

Hugo54, 2020-04-26:
No <3 But I wondering also how use 0 or 1.

```
var five = require("johnny-five");
var moment = require('moment');
var board = new five.Board({
    port: "COM3"
});

board.on("ready", function() {
    let start = moment()
    var lcd = new five.LCD({
        controller: "JHD1313M1"
    });
    lcd.bgColor("blue");
    lcd.useChar("heart")
    lcd.print(":heart:")

    lcd.useChar("1")
    lcd.print(":1:")
```

Thanks

dtex, 2020-04-26:
I'm surprised the special character is necessary. Have you tried just ```lcd.print("1");```?

Hugo54, 2020-04-26:
It's seems to be somme special char predefined by the library : http://johnny-five.io/api/lcd/
After of course if I want to print 1 I can ...

rwaldron, 2020-04-26:
I think that predefined character is for the special format shown. You can use plain ascii "1", but no control over how it looks.

amna625, 2020-04-26:
// this lcd.print(":heart:")

fivdi, 2020-04-28:
After invoking `useChar` the cursor needs to be positioned by invoking `cursor`, or at least this was the case in the past. See [here](https://github.com/rwaldron/johnny-five/issues/1478#issuecomment-404933363).