rwaldron/johnny-five #1677: Previously code was running but now it was showing only  Connected, Available and Repl initialized (as in pic ) and there is no more execution further.

Manish2208, 2020-04-26:
Previously code was running but now it was showing only  Connected, Available and Repl initialized (as in pic ) and there is no more execution further.
[image]
@dtex can u please tell what is happening here?

_Originally posted by @Manish2208 in https://github.com/rwaldron/johnny-five/issues/1218#issuecomment-619534258_

Manish2208, 2020-04-26:
here is the code......
let Five = require("johnny-five");
let board = new Five.Board();

board.on("ready", () => {

  board.io.accelStepperConfig({
    deviceNum: 0,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 5,
    motorPin2: 6,
    motorPin3: 7,
    motorPin4: 8,
    stepSize: board.io.STEPPER.STEP_SIZE.WHOLE
  });

  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,

    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });

  board.io.accelStepperSpeed(0, 400);
  board.io.accelStepperSpeed(1, 400);

  board.io.multiStepperConfig({
    groupNum: 0,
    devices: [0, 1]
  });

  board.io.multiStepperTo(0, [2000, 3000], () => {

    board.io.accelStepperReportPosition(0, value => {
      console.log(`Stepper 0 position: ${value}`);
    });

    board.io.accelStepperReportPosition(1, value => {
      console.log(`Stepper 1 position: ${value}`);
    });

  });

});

Manish2208, 2020-04-26:
here is the code...

> let Five = require("johnny-five");
let board = new Five.Board();

board.on("ready", () => {

  board.io.accelStepperConfig({
    deviceNum: 0,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 5,
    motorPin2: 6,
    motorPin3: 7,
    motorPin4: 8,
    stepSize: board.io.STEPPER.STEP_SIZE.WHOLE
  });

  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,

    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });

  board.io.accelStepperSpeed(0, 400);
  board.io.accelStepperSpeed(1, 400);

  board.io.multiStepperConfig({
    groupNum: 0,
    devices: [0, 1]
  });

  board.io.multiStepperTo(0, [2000, 3000], () => {

    board.io.accelStepperReportPosition(0, value => {
      console.log(`Stepper 0 position: ${value}`);
    });

    board.io.accelStepperReportPosition(1, value => {
      console.log(`Stepper 1 position: ${value}`);
    });

  });

});

Manish2208, 2020-04-26:
here is the code...
`let Five = require("johnny-five");
let board = new Five.Board();

board.on("ready", () => {

  board.io.accelStepperConfig({
    deviceNum: 0,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 5,
    motorPin2: 6,
    motorPin3: 7,
    motorPin4: 8,
    stepSize: board.io.STEPPER.STEP_SIZE.WHOLE
  });

  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,

    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });

  board.io.accelStepperSpeed(0, 400);
  board.io.accelStepperSpeed(1, 400);

  board.io.multiStepperConfig({
    groupNum: 0,
    devices: [0, 1]
  });

  board.io.multiStepperTo(0, [2000, 3000], () => {

    board.io.accelStepperReportPosition(0, value => {
      console.log(`Stepper 0 position: ${value}`);
    });

    board.io.accelStepperReportPosition(1, value => {
      console.log(`Stepper 1 position: ${value}`);
    });

  });

});

`

dtex, 2020-04-26:
First I'd confirm that everything is working by just hooking up an LED and blinking that. If it is, then I would assume you have the wrong version of firmata installed. Did you use configurableFirmata?

Manish2208, 2020-04-26:
`let Five = require("johnny-five");
let board = new Five.Board();

board.on("ready", () => {
  board.io.accelStepperConfig({
    deviceNum: 0,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 5,
    motorPin2: 6,
    motorPin3: 7,
    motorPin4: 8,
    stepSize: board.io.STEPPER.STEP_SIZE.WHOLE
  });
  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });
  board.io.accelStepperSpeed(0, 400);
  board.io.accelStepperSpeed(1, 400);

  board.io.multiStepperConfig({
    groupNum: 0,
    devices: [0, 1]
  });
  board.io.multiStepperTo(0, [2000, 3000], () => {
    board.io.accelStepperReportPosition(0, value => {
      console.log(`Stepper 0 position: ${value}`);
    });
    board.io.accelStepperReportPosition(1, value => {
      console.log(`Stepper 1 position: ${value}`);
    });
  });
});
_here is the code....._

Manish2208, 2020-04-26:
> First I'd confirm that everything is working by just hooking up an LED and blinking that. If it is, then I would assume you have the wrong version of firmata installed. Did you use configurableFirmata?

ya i'm using configurableFirmata 2.10.1 and firmata 2.2.0

Manish2208, 2020-04-26:
`const Board =require("firmata");

Board.requestPort((error, port) => {
  if (error) {
    console.log(error);
    return;
  }

  const board = new Board(port.path);

  board.on("ready", () => {
    const pin = 13;
    let state = 1;

    board.pinMode(pin, board.MODES.OUTPUT);

    setInterval(() => {
      board.digitalWrite(pin, (state ^= 1));
    }, 500);
  });
});`
this code is working properly , i found this from firmata.js documentation

Manish2208, 2020-04-26:
> `const Board =require("firmata");
>
> Board.requestPort((error, port) => {
> if (error) {
> console.log(error);
> return;
> }
>
> const board = new Board(port.path);
>
> board.on("ready", () => {
> const pin = 13;
> let [...]
it is blink code.

Manish2208, 2020-04-26:
> `const Board =require("firmata");
>
> Board.requestPort((error, port) => {
> if (error) {
> console.log(error);
> return;
> }
>
> const board = new Board(port.path);
>
> board.on("ready", () => {
> const pin = 13;
> let [...]

_it is blink code_

dtex, 2020-04-26:
Can you paste your configurableFirmata code into a gist?

Manish2208, 2020-04-26:
`/*
  Firmata is a generic protocol for communicating with microcontrollers
  from software on a host computer. It is intended to work with
  any host computer software package.

  To download a host software package, please clink on the following link
  to open the download page in your default browser.

  https://github.com/firmata/ConfigurableFirmata#firmata-client-libraries

  Copyright (C) 2006-2008 Hans-Christoph Steiner.  All rights reserved.
  Copyright (C) 2010-2011 Paul Stoffregen.  All rights reserved.
  Copyright (C) 2009 Shigeru Kobayashi.  All rights reserved.
  Copyright (C) 2013 Norbert Truchsess. All rights reserved.
  Copyright (C) 2014 Nicolas Panel. All rights reserved.
  Copyright (C) 2009-2017 Jeff Hoefs.  All rights reserved.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  See file LICENSE.txt for further informations on licensing terms.

  Last updated: September 16th, 2017
*/

/*
 README

 This is an example use of ConfigurableFirmata. The easiest way to create a configuration is to
 use http://firmatabuilder.com and select the communication transport and the firmata features
 to include and an Arduino sketch (.ino) file will be generated and downloaded automatically.

 To manually configure a sketch, copy this file and follow the instructions in the
 ETHERNET CONFIGURATION OPTION (if you want to use Ethernet instead of Serial/USB) and
 FIRMATA FEATURE CONFIGURATION sections in this file.
*/

#include "ConfigurableFirmata.h"

/*==============================================================================
 * ETHERNET CONFIGURATION OPTION
 *
 * By default Firmata uses the Serial-port (over USB) of the Arduino. ConfigurableFirmata may also
 * comunicate over ethernet using tcp/ip. To configure this sketch to use Ethernet instead of
 * Serial, uncomment the approprate includes for your particular hardware. See STEPS 1 - 5 below.
 * If you want to use Serial (over USB) then skip ahead to the FIRMATA FEATURE CONFIGURATION
 * section further down in this file.
 *
 * If you enable Ethernet, you will need a Firmata client library with a network transport that can
 * act as a server in order to establish a connection between ConfigurableFirmataEthernet and the
 * Firmata host application (your application).
 *
 * To use ConfigurableFirmata with Ethernet you will need to have one of the following
 * boards or shields:
 *
 *  - Arduino Ethernet shield (or clone)
 *  - Arduino Ethernet board (or clone)
 *  - Arduino Yun
 *
 *  If you are using an Arduino Ethernet shield you cannot use the following pins on
 *  the following boards. Firmata will ignore any requests to use these pins:
 *
 *  - Arduino Uno or other ATMega328 boards: (D4, D10, D11, D12, D13)
 *  - Arduino Mega: (D4, D10, D50, D51, D52, D53)
 *  - Arduino Leonardo: (D4, D10)
 *  - Arduino Due: (D4, D10)
 *  - Arduino Zero: (D4, D10)
 *
[... 344 lines cut ...]
    i2c.report();
#endif
#ifdef FirmataEncoder_h
    // report encoders positions if reporting enabled.
    encoder.report();
#endif
  }
#endif
#ifdef StepperFirmata_h
  stepper.update();
#endif
#ifdef AccelStepperFirmata_h
  accelStepper.update();
#endif
#ifdef SerialFirmata_h
  serial.update();
#endif

#if defined NETWORK_FIRMATA && !defined local_ip &&!defined _YUN_CLIENT_H_
  // only necessary when using DHCP, ensures local IP is updated appropriately if it changes
  if (Ethernet.maintain()) {
    stream.maintain(Ethernet.localIP());
  }
#endif
}`

Manish2208, 2020-04-26:
> Can you paste your configurableFirmata code into a gist?

 _i am using normal configurableFirmata example_
[image]

Manish2208, 2020-04-26:
<script src="https://gist.github.com/Manish2208/470ca9372c14636f53e25ec1e117d518.js"></script>
here is the gist

Manish2208, 2020-04-26:
https://gist.github.com/Manish2208/470ca9372c14636f53e25ec1e117d518
_here is the gist of ConfigurableFirmata_

dtex, 2020-04-26:
It looks like you haven't commented out any of the optional parts. You may be running out of memory. Either comment out the things you don't need, or use http://firmatabuilder.com/ to build a configurableFirmata that's made just for you.

Manish2208, 2020-04-26:
Now I had build new firmata using firmatabuilder.com
[image]
https://gist.github.com/Manish2208/6b7d09e2d1ea106ace140475e6c9e9af

Manish2208, 2020-04-26:
thanks a lot, now above code working

Manish2208, 2020-04-26:
Did u know any algorithm or code to run 2 stepper motor for linear motion in x and y direction using web page?

dtex, 2020-04-26:
https://github.com/firmata/firmata.js/blob/master/packages/firmata.js/examples/stepper-multi.js

Manish2208, 2020-04-30:
how should I change the direction of stepper motor in between code ?

dtex, 2020-05-29:
If you are calling ```step()```, you can pass a negative number. I like to use ```to()``` and pass a specific target position. Let accelStepper handle keeping track of the number of steps you need to take and the direction you need to go.

Manish2208, 2020-05-30:
> If you are calling `step()`, you can pass a negative number. I like to use `to()` and pass a specific target position. Let accelStepper handle keeping track of the number of steps you need to take and the direction you need to go. 

Can u explain `to()` using example? I have already written code for `step()` which uses button to change it direction and it runs successfully .But now the problem is how to use `while loop` in johnny five?
```
let Five = require("johnny-five"),
button;
var loop = require('node-while-loop');
let board = new Five.Board();
var x=-12000;
board.on("ready", () => {
  button = new Five.Button(2);

  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });
 
 // board.io.accelStepperSpeed(0, 400);
  board.io.accelStepperSpeed(1, 1200);
 
  button.on("down", function() {
    console.log("down");
    x=12000;
    board.io.accelStepperStep(1,x);
  });
  board.io.accelStepperStep(1,x);
});
```

Manish2208, 2020-05-30:
> If you are calling `step()`, you can pass a negative number. I like to use `to()` and pass a specific target position. Let accelStepper handle keeping track of the number of steps you need to take and the direction you need to go.

Can u explain `to()` using example? I have already written code for `step()` which uses button to change it direction and it runs successfully .

```
let Five = require("johnny-five"),
button;
var loop = require('node-while-loop');
let board = new Five.Board();
var x=-12000;
board.on("ready", () => {
  button = new Five.Button(2);

  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });
 
 // board.io.accelStepperSpeed(0, 400);
  board.io.accelStepperSpeed(1, 1200);
 
  button.on("down", function() {
    console.log("down");
    x=12000;
    board.io.accelStepperStep(1,x);
  });
  board.io.accelStepperStep(1,x);
});
```

dtex, 2020-05-30:
````js
let Five = require("johnny-five"),
button;
var loop = require('node-while-loop');
let board = new Five.Board();
var x=-12000;
board.on("ready", () => {
  button = new Five.Button(2);

  board.io.accelStepperConfig({
    deviceNum: 1,
    type: board.io.STEPPER.TYPE.FOUR_WIRE,
    motorPin1: 9,
    motorPin2: 10,
    motorPin3: 11,
    motorPin4: 12,
    stepSize: board.io.STEPPER.STEP_SIZE.HALF
  });
 
 board.io.accelStepperSpeed(1, 400);
 
  button.on("down", function() {
    console.log("down");
    board.io.accelStepperTo(1,12000);
  });

  button.on("up", function() {
    console.log("up");
    board.io.accelStepperTo(1,0);
  });

});
````

BTW, 1200 is too high a step speed for accelStepper.