rwaldron/johnny-five #1678: Could not locate the bindings file. Tried:

Vinaayakk97, 2020-04-29:
I have been working on a project : How to control arduino wirelessly with WebSockets and Johnny-five.
I m referring to 
https://www.bracketsacademy.com/2018/10/01/how-to-control-arduino-wirelessly-with-web-sockets-and-johnny-five/

I get this error 
**Error: Could not locate the bindings file. Tried:**

server.js

`const express = require('express');
const app = express();
const port = 3000;

var five = require("johnny-five");
const WebSocket = require('ws');
const wss = new WebSocket.Server({port:8080});

app.listen(port, () => { console.log("we are live") })
app.use('/color',express.static('public'));

five.Board().on('ready',function(){
    var led = new five.Led.RGB({
        pins:{
            red:6,
            green:5,
            blue:3
        }
    });
    led.on();
    led.color('#FF0000');
    wss.on('connection',function(ws,req){
        console.log('connected');
        ws.on('message',function(data){
            led.color('#' + data);
        })
    })
})`

package.json

`{
  "name": "rgb",
  "version": "1.0.0",
  "description": "RGB Lights Example.",
  "main": "server.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1",
    "start": "node server.js"
  },
  "author": "",
  "license": "ISC",
  "dependencies": {
    "express": "^4.17.1",
    "johnny-five": "^1.4.0",
    "serialport": "^8.0.7",
    "ws": "^7.2.5"
  }
}
`

fivdi, 2020-04-30:
>I get this error
>#Error: Could not locate the bindings file. Tried:#

fivdi, 2020-04-30:
>I get this error
>Error: Could not locate the bindings file. Tried:

This error message usually indicates that a native module has not been installed correctly. Unfortunately only the error message rather than the complete error stack has been provided so it's not 100% clear which native module is causing the issue. In this case it's likely to be `serialport`.

Please do the following
* Verify that the johnny-five and it's dependencies have been installed successfully
* Ensure that the "Hello World!" example shown on http://johnny-five.io/ that blinks an LED functions correctly
* If there is still an error message post the complete error stack