rwaldron/johnny-five #1682: Uncaught 1-Wire timeout

misterzirillo, 2020-05-15:
I'm using some DS18B20 1wire devices via `var sensor = new five.Thermometer({ controller: "DS18B20" })`. The devices work properly however when the board is disconnected (in software or otherwise) node quits with the following error:

```
./node_modules/firmata-io/lib/firmata.js:1632
      callback(new Error("1-Wire device read timeout - are you running ConfigurableFirmata?"));
               ^
Error: 1-Wire device read timeout - are you running ConfigurableFirmata?
    at Timeout._onTimeout (./node_modules/firmata-io/lib/firmata.js:1632:16)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
```

I have on "error" handlers for both the board and sensor objects but they don't make a difference. 

I've also looked at [the line in question](https://github.com/firmata/firmata.js/blob/a94b8cceaee6e08d5c60deeafed8788b345dcf75/packages/firmata-io/lib/firmata.js#L1632). Looking for  `sendOneWireWriteAndRead` in this codebase I came across [this line](https://github.com/rwaldron/johnny-five/blob/d278bc46b506277f088412ef556f7659e61cfa47/lib/thermometer.js#L102) where an error might not be accounted for.

Am I right?

If it's as simple as this then I can submit a PR. Otherwise, I'm having trouble understanding how to get from `five.Thermometer` to `sendOneWireWriteAndRead`.

```javascript
if (err) {
  this.emit("error", err);
  return;
}
```

misterzirillo, 2020-05-15:
Looking through the open issues I can see this is possibly a duplicate of #1665. However I've got no issues using one or multiple thermometers for extended periods of time. I only see the timeout after the board is disconnected.

saksholm, 2020-05-26:
@misterzirillo I struggled a long time with that DS18B20 '1-Wire device read timeout' problem on #1665 . If more than 2 sensors were attached it crashes after some ~20mins. I need at least 14 for my project and stable reading. I never found a reason for that and solved this with a backpack. [My dirty hack](https://github.com/saksholm/i2c-backpack) not cleaned and really messy. That sends 2bytes per temperature reading (integers which are multiplied by 100). Total 28bytes (32maximum per time)

Attached sensors to Uno board and created a program for that. Now it communicates via I2C, reads all sensor data once a second, and then I can do whatever I want.

Sounds stupid hack for me but I have lived with that unless I or someone else found an easy solution for this timeout problem.

closed 2022-01-09 by stale[bot]