rwaldron/johnny-five #1683: Keyboard.write() alternative in Johnny-five? (Sending keystroke with single switch button)

w90, 2020-05-17:
Is there a way to send keystroke to my PC with J5 using a single Switch hardware element on my breadboard?
I know about http://johnny-five.io/api/keypad/, the thing is - I'd like to program a single switch button to send the keystroke.
Thanks for any help!

dtex, 2020-05-17:
Well, there's this [module](https://www.npmjs.com/package/node-key-sender) but it requires a JRE. What app/device will be listening for the keystroke? Maybe there is a cleaner solution.

w90, 2020-05-17:
I plan for arduino to gather keystrokes from multiple switch-buttons and then my react.js-app should catch them and handle all the sync/async logic. Thanks for a possible way to do this!

w90, 2020-05-17:
I have found this npm package, works like a charm!
https://www.npmjs.com/package/robotjs

dtex, 2020-05-18:
That's great. Good luck on your project.

closed 2020-05-18 by dtex