rwaldron/johnny-five #1684: no funciona johnny-five

camora28, 2020-05-25:
hola buen dia, estoy inicializando con johnny five y no me funciona estoy ejecutando las indicaciones dadas guia de inicio  , tengo ; windows 10 64 :npm --version 6.14.5 : node `--versionv13.10.1
[image]
[image]
[image]

`
----------------------------------
me sale el siguiente error 
PS E:\CURSO-IOT\ARDUINO\ARDUINO_JOHNNY-FIVE\jhonnyFive_HolaMundo> node index.js
internal/modules/cjs/loader.js:979
  throw err;
  ^

Error: Cannot find module '@serialport/bindings'
Require stack:

dtex, 2020-05-25:
Hola, este problema es con una de las dependencias de Johnny-Five. Sospecho que la instalación de npm no se ejecutó correctamente. Lo primero que intentaría es reinstalar SerialPort. Ejecute ```npm install serialport```. Observe la salida muy de cerca y vea si hay algún error o advertencia. Si no, intente ejecutar su programa nuevamente y vea si algo cambia.

Pido disculpas por mi español. Estoy dependiendo de Google Translate.

camora28, 2020-05-29:
hola gracias por la pronta respuesta, segui las indicaciones pero persiste el problema creo que el problema es con npm voy a reinstalarlo por que ocurre problemas con otras aplicaciones que estoy tratando de desarrollar y me arroja los mismos problemas de que falta el modulo.

[image]
[image]
[image]