rwaldron/johnny-five #1687: Support for generic flame sensors

meerasndr, 2020-06-16:
Does Johnny-Five support generic flame sensors? The documentation and examples mention the Intel Edison + Grove sensor, but I don't get this in my country. Would something like KY-026 work? Thanks.

dtex, 2020-06-16:
@meerasndr The KY-026's I found on the web have two outputs. One digital (if a flame is detected) and one analog (0-1023 depending on flame intensity). The Intel Edison + Grove sensor example will work for the digital output (or you could just use Switch) and then you can set up a second, generic sensor instance to read the analog value.

meerasndr, 2020-06-17:
Thanks @dtex . I'll play around with this and post my findings here.

meerasndr, 2020-08-25:
Yes please!

closed 2020-08-25 by meerasndr