rwaldron/johnny-five #1689: lcd.createChar() not documented?

Caleb-Irwin, 2020-06-27:
I noticed that this is not documented in the lcd api page and I don't know why. I could not find it in any of the documentation, but I typed it into the repl and there was a valid function. It takes a name (String) and an array of hex numbers. You can some of those arrays in the /lib/lcd-chars.js file. Am I looking in the wrong place or is this not documented at all?

Caleb-Irwin, 2020-08-23:
Yes that's perfect

dtex, 2020-08-23:
Cool. Next time the site gets published, the docs will be updated there too.

Thanks!

closed 2020-08-23 by dtex

Caleb-Irwin, 2020-08-23:
go [here](https://github.com/rwaldron/johnny-five/blob/master/lib/lcd-chars.js) to see what the the original characters are defined as.