rwaldron/johnny-five #1691: Query not returning callback

fencer, 2020-07-04:
when querying the pin state on beaglebone green wireless using the example code below the callback function is never executed.

```
var pin = new five.Pin(13);

pin.query(function(state) {
  console.log(state);
});
```

(using the same pin will fire the high / low event so it's not a hardware issue)

dtex, 2020-07-04:
```query()``` just wraps the IO plugin's ```queryPinState()``` method. I don't have a Beaglebone to try and recreate, but a quick test to see if this is a Johnny-Five issue, or something that should be asked over in beaglebone-io could be:
````js
// Let J5 configure the pin
var pin = new five.Pin("P8_29");

// Query beaglebone-io directly (assumes your board is named "board")
board.io.queryPinState("P8-29", function(state) {
  console.log(state);
});
````

dtex, 2020-07-04:
This [test file](https://github.com/julianduque/beaglebone-io/blob/56779176904acdd95932bff5318ee35edfc9173f/test/not-implemented.js) suggests to me that queryPinState has not been implemented in beaglebone-io.

fencer, 2020-07-04:
Wow! Wouldn't have expected this! What other alternative methods I can use instead?

dtex, 2020-07-04:
Tell me what you are trying to do. Typically J5 programs do not need pin.query.

fencer, 2020-07-04:
My use case isn't the typical J5 implementation but what i have is a number of analogue sensor that I need to read the value of. Along with those I have other circuitry that will send a digital high when the state changes. To display these readings I have a websocket application. I am currently changing the values onthe page by listening to the high / low events but when the server application is loaded the value is not updated untill the pin state is toggled (which will fire the high/low event). To solve this I would need to query the pin state on application start

dtex, 2020-07-04:
Can you just read the pin value property after calling ```read()```?
````js
console.log(pin.value);
````

fivdi, 2020-07-04:
If a Pin is created and its mode is set to `INPUT` or `ANALOG` then its `value` property will be updated at regular intervals by Johnny-Five automatically. However, there may be issues if the `value` is accessed before Johnny-Five has a chance to update it for the first time. If the `value` is accessed before Johnny-Five has updated it for the first time then Johnny-Five will say that the value is 0 although it may be 1. It's possible to workaround this issue by adding a short delay before accessing `value` for the first time. The below program attempts to show this.

```js
var five = require('johnny-five');
var BeagleBone = require('beaglebone-io');

var board = new five.Board({
  io: new BeagleBone()
});

board.on('ready', function() {
  var pin = new five.Pin({
    pin: 'P8_29',
    mode: this.MODES.INPUT
  });

  console.log('pin.value (may print incorrect value): ' + pin.value);
  console.log('readSync() HACK (will print correct value): ' + board.io._pins[board.io.normalize('P8_29')].gpio.readSync());
  setInterval(function () {
    console.log('pin.value (will print correct value): ' + pin.value);
  }, 1000);
});
```

This program also shows a terrible hack that relies on implementation details of beaglebone-io that could also be used to workaround the issue.

closed 2022-04-17 by stale[bot]