rwaldron/johnny-five #1693: Problem to execute a Loop with Johnny-five

Daltsz, 2020-08-02:
Hello, I'm with a problem, that in my code I'm executing a Switch/Case inside of a Loop to pass by an array with the purpose of blink a led many times,  but when passing by the array, the led blink in the first position of the array and in the next positions don't happen anything. I would like some help with this problem because I'm desperate

Daltsz, 2020-08-02:
Inside of array, I'm passing these strings: 
[ 'Blink_Led', 'Blink_Led', ' ' ]

below it's my code:

app.post("/led-blink", async (req, res) =>{

    let i = 0 

        try{
            
            for (i=0 ; i< (req.body.length-1); i++){

                console.log(req.body[i]);

                    switch(req.body[i]){
                        
                        
                       
                        case 'Blink_Led':
                            const led = new five.Led(13);

                            await led.blink(1000);
                            //res.json({message: 'sucess!!'});
                            console.log('Passed Here')
                            await board.wait(2000, async () =>{

                               await led.off().stop();
                               
                                
                                res.writeContinue();
                                
                            });
                            
                            break;

                        default:
                            res.json('NÃO PASSOU NENHUM PARAMETRO VALIDO');
                              
                    }
                    
            }  

        } catch (error) {

            res.status(400).json({message: error.message});
            
        }
        //
      
    
    res.json({message: 'Sucess!!!'});

});

dtex, 2020-08-02:
One problem is that you are re-instantiating the LED on every pass through the loop. This line:
````js
const led = new five.Led(13);
````
should never be run more than once. I would put it outside the app.post handler

Daltsz, 2020-08-02:
I made this, put this line outside of the loop, but the led blinks a unique time, in the first string of the array

dtex, 2020-08-02:
You are not waiting for the first blink to finish before calling the second one & those LED and Board methods are not asynchronous.

Daltsz, 2020-08-02:
Oh, understand but how to accomplish this? Wait for the first blink finish to call the next blink?

dtex, 2020-08-02:
There are a lot of ways to solve this, but I would probably append the incoming array to a queue and have a ```processqueue()``` method that gets only called once by your led-blink route with req.body as the only param. Inside processqueue, I would append the incoming param to your queue, shift out the first value of the array and switch on that, and then call ```processqueue()``` again from the board.wait callback with the returned value from your shift call. You will have to keep track of whether processQueue is already running, but that's easy with an ```isRunning``` boolean.

Subsequent requests for "/led-Blink" could then append to your queue if that's how you want to handle it.

Daltsz, 2020-08-02:
oh ok, I'll try this, thanks