rwaldron/johnny-five #1694: Using digital Arduino-Uno pin as PWM doesn't work

Jogifromearth, 2020-08-04:
I try to control the speed of a fan, that uses PWM-signals. Therefor I use the 'Pin' class and I tried to set the mode to 3, which stands for PWM based on the johnny-five API. Unfortunately the mode stays at 1 (output), when I log the pin state in the console. 
As soon as this works, I like to send values from 0 to 255 via the write() method, as it is usual for PWM-Pin in Arduino-Programs.

`const five = require("johnny-five");
  const board = new five.Board();

board.on("ready", function () {

    let pin1 = new five.Pin({
        pin: 11,
        mode: 3
    });

    pin1.write(10);

    pin1.query((state) => console.log(state));
    // Logged state-object:
    // {
    //     supportedModes: [ 0, 1, 3, 4, 8 ],
    //     mode: 1,
    //     value: 10,
    //     report: 1,
    //     analogChannel: 127,
    //     state: 1
    //   }
});`

dtex, 2020-08-20:
I'm seeing similar problems with this code:

````js
const { Board, Led } = require("johnny-five");
const board = new Board();

board.on("ready", function() {
  const led = new Led(12);
  led.pulse();
  console.log(led);
});
````

Pretty sure this is going to be traced down to firmata, and not Johnny-Five. I'm going to create a test case that excludes J5.

dtex, 2020-08-20:
Though I see this user is on an ESP8266 so I assume 9 on there is fine for PWM so probably a different issue.

Jogifromearth, 2020-09-09:
Thank you! I used the "pinMode" and the "pwmWrite"-Method and it also worked:

`const five = require("johnny-five");
  const board = new five.Board();

board.on("ready", function () {

    board.pinMode(11, Pin.PWM);
    board.pwmWrite(11, 255); //Pin, Value (0-255);

}`

dtex, 2020-09-09:
Excellent!

closed 2020-09-09 by dtex