rwaldron/johnny-five #1697: No data from Adafruit BNO055

pauldb126, 2020-08-08:
Hi, first of all, really appreciate Johnny-five. 
Everything I have tried works just as I would expect, apart from the Adafruit BNO055 IMU.

I have tried it hooked up to an Arduino Uno and a Nano. I am running Node on a Raspberry Pi and connecting to the Arduino via USB. Everything seems fine. Board initializes. I have an Adafruit Ultimate GPS hooked up aswell and that gives readings.

In order to test the BNO055 board and Arduino, I unplugged the Raspberry Pi. I then connected the Arduino board, with BNO055 attached, to my PC and used the Ardiuno IDE to install and run their example programs. All work fine, readings as I would expect. This indicates to me that the pin connections between the Arduino and the BNO055 are correct.

So, returning the Arduino connection to the Raspberry Pi, I tried specifying the Port used for the board in order that I could specify different baud rates. However, despite following the examples. That did not seem to work either, I get an error related to the board pins.

Any help would be greatly appreciated. Many thanks, Paul

pauldb126, 2020-08-08:
It's just occurred to me that maybe I cannot have both the BNO055 and GPS breakouts hooked up to the same Arduino controller. I will experiment more.
In the meantime, I have found someone that seems to experience the same issue:

https://forum.arduino.cc/index.php?topic=494127.0

pauldb126, 2020-08-09:
> 
> 
> It's just occurred to me that maybe I cannot have both the BNO055 and GPS breakouts hooked up to the same Arduino controller. I will experiment more.
> In the meantime, I have found someone that seems to experience the same issue:
> 
> https://forum.arduino.cc/index.php?topic=494127.0

Tried again without the GPS breakout hooked up. Same result.
There's either an issue with my setup or Johnny-Five is failing to get readings from the BNO055. As I said in my original post, the breakout and Arduino board respond perfectly when connected to the Arduino IDE.

pauldb126, 2020-08-11:
The example eg/imu-bno055.js does not work for me either.

dtex, 2020-08-11:
Just curious, have you tried using the BNO055 with Johnny-Five running on your computer instead of the Pi?

pauldb126, 2020-08-11:
> Just curious, have you tried using the BNO055 with Johnny-Five running on your computer instead of the Pi?

Thanks for the suggestion. I will try that.

One thing I noticed during installation of npm packages on the pi, there were a number of warnings when installing 'serialport' but none seemed fatal to me.

Anyhow, I will try from the PC. Thanks.

pauldb126, 2020-08-12:
> 
> 
> Just curious, have you tried using the BNO055 with Johnny-Five running on your computer instead of the Pi?

Tried it. Same result. All readings giving zero values. That is not the case when I use the Arduino IDE and run their sample programs which makes me think I am using the right pin connections.

If the baud rate on the USB port was wrong, would I see an error? Also, as I have mentioned, I have a GPS breakout attached to the same Arduino controller and I get results from that.

pauldb126, 2020-08-16:
I retried connected to my PC rather than the Pi and I managed to get it to work by moving the BNO055 around in order to calibrate it.
However, it still would not work when connected to the Pi. However, I connected it to a separate Arduino Nano - i.e. not the same one that the GPS was connected to. Then it worked. However, it was sensitive to the USB port on the Pi to which the board was connected.
So, if there is an issue - then it's that two breakouts cannot work off the same Arduino Nano - but that might not be a Johnny-Five issue. There is also a question about the sensitivity of the USB port used. Again, that might be an issue with the Pi or with me, not necessarily with Johnny Five (I will try specifying the port for the board).
In any case, the reason why I created this issue in the first place no longer seems valid so I will close it.

closed 2020-08-16 by pauldb126