rwaldron/johnny-five #1701: Define targeted board

Yvanovitch, 2020-08-20:
Hi,
I have 2 or more boards connected to my device (RBPI)
In the doc, there is no reference to a "board" option that would allow me to define which board is targeted in the Pin and the Button classes contructors option.
Is it an option that is there but not referred in the doc, or is it not an option ?

Thanks

dtex, 2020-08-20:
It's an option. Here's an example:

http://johnny-five.io/examples/board-multi/

Can I ask a quick question? There have been a lot of people recently wanting to use the Pin class directly which is kinda unusual. Is there a class or group of some kind that is learning to use J5 together and learning the Pin class?

Yvanovitch, 2020-08-20:
Thanks
Is there a reason why it's not referred in the Pin class's doc ?
Not that I know of, I'm independent

dtex, 2020-08-20:
I think it's probably just that the Pin class doesn't see much action so it's been overlooked all these years. I've added it the wiki and next time the johnny-five site is published it will be updated there as well.