rwaldron/johnny-five #1702: Servo: Detach / stop update?

saschagehlich, 2020-08-30:
Hey,

is there a way to detach the servo so that it is movable by hand? I think Arduino's `Servo` library has a `.detach()` method for that...?

jcmertz, 2021-02-10:
I could also really use this. I have a mechanism where servos are pulling open a spring loaded door, and then returning to their closed position. I am struggling to get the servos to line up nicely with the door in the closed position with the result being they are sometimes left under some load and end up running continuously to try to hold their position, which I don't need them to do. I'd like to be able to stop sending a PWM signal and tell them to "relax", but I can't seem to find a way to do that with the servo api in johnny-five. I have already tried stop(), but I think that just tells it to hold its current position. Anyone have any ideas?

dtex, 2021-02-10:
I don't think a servo will relax unless you remove power. No pulse would just send it to the low end of its range.

closed 2022-04-27 by stale[bot]