rwaldron/johnny-five #1705: Using serial port to recieve sensor data from Atlas scientific Ezo Ph

Bohemus307, 2020-10-07:
Is there a way currently to get data from the serial port (usb) on a raspberry pi 3? I have been playing with serial read but no success yet.I am trying to get data from a atlas scientific ph sensor that connects to the usb port. Any advice or direction on this would be welcomed.

Prath06, 2020-10-28:
You need the following few Hardware:
1.	Electrically Isolated USB EZO Carrier Board: for BNC to USB
2.	USB-A to USB mini B cable
In the Python Script, Add the following:
import serial
ser = serial.Serial('/dev/ttyACMn', 9600)
where ttyACMn, n may be 0,1,2,...

Bohemus307, 2020-10-29:
Thanks for the reply I appreciate it. I have those things and can get the sensor running using python script I am just uncertain as to how I can apply that to johnny-five and utilize its sensor component to connect and retrieve data from the sensor?

closed 2022-04-17 by stale[bot]