rwaldron/johnny-five #1706: Writing to PWM pins has no effect or doesn't run at all

n00ne1mportant, 2020-10-12:
Hello!
I'm trying to write a value to PWM pin but no use. Here's the code:
```js
  newBoard.on('ready', function () {
    const led = new five.Led(13)
    led.blink(500)
    const pwmPin = new five.Pin({
      pin: 9,
      type: 'analog'
    })
    pwmPin.query(state => console.log(state))
    pwmPin.write(100)
  })
}
```

Pin 9 is connected to a LED through resistor and to the ground. I'm trying to light the LED as specific brightness using analogWrite. It just remains off no matter what value I put.
So my circle is just this:
PIN9 --> LED --> Resistor --> GROUND
It works correctly on ArduinoIDE sample code (Fade) so I'm sure connection is correct (not a lot of space for errors anyway)
**Symptoms:**
Query event not firing if type is "analog", if "digital" it does and LED lights at full brightness, which is what I don't want.
Tried using different methods of pin control (boradInstance.io.analogWrite etc.).
`mode` property is useless as far as I can see, as it make no difference what mode the pin is when used as it's always inferred, correct?  
PWM mode doesn't work and mode=2 (analog) gives error right away (null reference). Only thing that does work is "digital" write both 1 and 0 but analogWrite just won't fire (LED is off all the time no matter what I try unless switched to "digital").

I didn't find any example of PWM usage anywhere in the docs or analogWrite for that matter...
Since `.query` won't fire, I'm guessing the write call never runs.  Constructor gets stuck somewhere... maybe? I don't know.
I'm not sure if I'm doing smtg wrong :/

Thank you for this amazing lib BTW,  I've build web controlled RC car (I'm using my phone for that) and it works flawlessly.
The only problem I'm having is PWM speed control. It runs at full speed or not at all. The LED setup in this issue is me testing my sanity. No conclusions so far! :+1: 

Thanks in advance!

dtex, 2020-10-12:
I believe pin is digital only.

This will work:

````js
  newBoard.on('ready', function () {
    const led = new five.Led(13)
    led.blink(500)
    const pwmLed = new five.Led(9);
    led.brightness(100);
  })
}
````

Also, there is an [ESC class](http://johnny-five.io/api/esc/) for speed controllers.

n00ne1mportant, 2020-10-12:
Thanks for the reply!
I will test it out tonight and hopefully close the issue. :+1: 

However... If the Pin is digital only there are corrections to be made in the docs. Also, I read the code for Pin, and it doesn't look digital only to me. Take a look:
```js
Pin.write = (pin, val) => {
  const state = priv.get(pin);

  state.value = val;

  // Set the correct mode (OUTPUT)
  // This will only set if it needs to be set, otherwise a no-op
  pin.mode = modes.OUTPUT;

  // Create the correct type of write command
  pin.io[`${pin.type}Write`](pin.addr, val);

  pin.emit("write", null, val);
};
```
This is from [here](https://github.com/rwaldron/johnny-five/blob/master/lib/pin.js#L223).

This line is interesting one: ```pin.io[`${pin.type}Write`](pin.addr, val);```  
Looks to me it calls analog or digital write depending on type. Mode seams useless however, since it will be set to OUTPUT in any case. 

If this is not intended, I can fix it up in PR but probably better to do so from a separate issue.

Thanks again for replaying! <3

dtex, 2020-10-12:
I'll test too. A similar issue came up recently, and I recall tracing it to Pin not handling analog write properly so I'm just going off memory.

n00ne1mportant, 2020-10-12:
Code you gave me works fine for PWM.
I just defined my motor PWM pin as a LED pin and it worked as expected (using brightness function). This is not how it's supposed to work however.
Before you ask, I tried the Motor.js module but I couldn't get it to work properly. I'm using the [Dual motor driver](https://www.pololu.com/product/2511) and there is even a config for it but it won't run in reverse, only forward. Manual pin assignment also didn't work as expected. So I did this as a test and this works:
```js
    const pin1 = new five.Pin({ pin: process.env.DCMOTOR_PIN_D1 })
    const pin2 = new five.Pin({ pin: process.env.DCMOTOR_PIN_D2 })
    const pin3 = new five.Led({ pin: process.env.DCMOTOR_PIN_PWM })
    const speed = 200
    // run forward 1 sec
    pin1.high()
    pin2.low()
    pin3.brightness(speed)
    // stop!
    setTimeout(() => {
      pin1.high()
      pin2.low()
      pin3.brightness(0)
    }, 1000)
    // run reverse 1 sec
    setTimeout(() => {
      pin1.low()
      pin2.high()
      pin3.brightness(speed)
    }, 2000)
    // Done!
    setTimeout(() => {
      pin1.high()
      pin2.low()
      pin3.brightness(0)
    }, 3000)
``` 
Tried changing speed, works as it should! So thanks for that :+1: 
There should be `Pin.writePwm` or functioning `analogWrite` or something... Using LED is a workaround imho

You can close now, thanks again!

dtex, 2020-10-12:
That looks like a different board (Pololu has more than one based on that chip).

Try this:

````js
const motor = new five.Motor([process.env.DCMOTOR_PIN_PWM, process.env.DCMOTOR_PIN_D2, process.env.DCMOTOR_PIN_D2]);
motor.forward(speed);
setTimeout(() => {
  motor.reverse(speed);
}, 1000)
````

n00ne1mportant, 2020-10-13:
Your motor code worked. I tired using pins object, and that didn't work...

Thank you again!
I'll close now