rwaldron/johnny-five #1707: Servo interfering with AnalogWrite on other pins

n00ne1mportant, 2020-10-18:
Hello! It's me again.

I'm having the following code:
```js
const five = require('johnny-five')
const motorAction = require('./constants/motor')
const servoAction = require('./constants/servo')
const motor = require('./customMotor')

const servo = new five.Servo({
  pin: process.env.SERVO_PIN,
  startAt: process.env.SERVO_BALANCE_ANGLE,
  range: [0, 180],
  center: true
})

function act (type, data) {
  switch (type) {
    case motorAction.FORWARD:
      motor.forward(data)
      break
    case motorAction.REVERSE:
      motor.reverse(data)
      break
    case motorAction.SWITCH:
      motor.switchSide()
      break
    case motorAction.STOP:
      motor.stop()
      break
    case servoAction.RIGHT:
      servo.to(parseInt(servo.startAt) + parseInt(data))
      break
    case servoAction.LEFT:
      servo.to(servo.startAt - data)
      break
    case servoAction.RESET:
      servo.to(servo.startAt)
  }
}
```
And this is my CustomMotor.js:
```js
const five = require('johnny-five')

const dir = new five.Pin(7)
const cdir = new five.Pin(8)
const pwm = new five.Led(9)

const cMotor = {
  forward: (value) => {
    pwm.brightness(value)
    dir.high()
    cdir.low()
  },

  stop: () => {
    // pwm.brightness(0)
    dir.low()
    cdir.low()
  },

  reverse: (value) => {
    pwm.brightness(value)
    cdir.high()
    dir.low()
  }
}

module.exports = cMotor
```

Controller is called by keyboard interface or by joystick via web socket.
I have connected LEDs on the breadboard to pins 7,8,9. I can 100% conclude that PWM pin is NOT working if Servo class is defined. However, if I comment it out and unplug/plug the board (RESET won't help) it begins to work fine. 

Creation of Servo kills the PWM. Can you please confirm this?
I have lost 10+ hours until I found this bug. All I see is PWM not firing, but couldn't figure out why. This is 100% reproducible

dtex, 2020-10-18:
Arduino Uno? What pin is the Servo on?

n00ne1mportant, 2020-10-18:
Yes, it's Uno, servo on pin 11
Motor on pin 9

dtex, 2020-10-18:
Try the servo down on pin 5. I think it's a shared timer issue.

n00ne1mportant, 2020-10-18:
I changed it to pin number five and left everything else unchanged, still the same problem.

If it helps, I changed the order of the creation so I create and use the motor first. The PWM LED on the board goes off, then it blinks and stays off after that. No longer responding. If Servo is not used, it stays on and behaves as expected.

n00ne1mportant, 2020-10-19:
I have found the problem.
Even the official Arduino library for servo has the same issue.

When servo is attached pins 9 and 10 can no longer be used as PWM pins. Attaching PWM to pin 5 seams to work fine even with servo on 11.

This is a problem for me doe since I'm using motor driver shield and it defaults on pins 9 and 10 for motor drivers. I'm figuring out workaround as we speak and then I will close the issue.
I think docs should be updated to highlight this problem in Servo section. I'll send you the PR later today.

References:
https://forum.arduino.cc/index.php?topic=64845.0

https://www.arduino.cc/reference/en/libraries/servo/

^Usage section

Prath06, 2020-10-27:
I faced such problems due to broken jumper wires.
Check the jumper wires or soldering points.

n00ne1mportant, 2020-10-28:
Hi, I fixed it.
Pins 9 and 10 are not working when servo is connected, it's not bad wiring or anything like that.
The way I fixed it is hackish. Since I need pin 9 to make the motor shield run and it won't work with servo, I soldered the pin on the upside of the shield so it doesn't connect to pin 9. Then I just connected that pin to PWM pin 3.

Picture is a thousand words:
[1000 words](https://prnt.sc/v8ul0q)

Pull request is opened

closed 2022-04-17 by stale[bot]