rwaldron/johnny-five #1709: Accelerometer: LIS3DH / LIS3DHTR

ChristophLabacher, 2020-10-25:
I wonder if the LIS3DH (or the identically constructed LIS3DHTR) accelerometer is supported. According to the [Accelerometer page](http://johnny-five.io/api/accelerometer/) it is not, but I did find this example on the website: [Accelerometer - LIS3DH](http://johnny-five.io/examples/accelerometer-LIS3DH/). I can also see that there is some code for it in the library. However when I try to initialize it my app crashes.

I did not find any past issues related to this, so I am wondering if there is full or partial support for this, or if there is any way I (with extremely limited knowledge of the inner workings of J5) can make any contribution to move it along?

rwaldron, 2020-10-25:
Would you mind sharing the crash output? Thanks!

ChristophLabacher, 2020-10-25:
I get `TypeError: Cannot read property 'io' of null`:

[image]

I initialize the component like this:

```
    this.j5Object = new five.Accelerometer({
      controller: "LIS3DH",
      board: board,
    });
```

When I call `console.log(board.io);` right before that I get a value, so the board seems to be initialized.

rwaldron, 2020-10-26:
Can you share your code?

rwaldron, 2020-10-26:
Also, I'm curious what you're running this code in? That output is unfamiliar to me.

ChristophLabacher, 2020-10-27:
@rwaldron I am running it as part of an electron app. The interface is bootstrapped with [Create React App](https://github.com/facebook/create-react-app), which provides the styling for the error page if I’m not mistaken. 

Our code is quite intricate (I am working on the app [blokdots](https://blokdots.com)), so it’s hard to single out the component. I will try to create a smaller reproducible example and report back later.

I started passing in a reference to `board` to the constructor functions for the components, because simpler components (like LED) were not working without it. I don’t think the Accelerometer class accepts this option – maybe this could be related to my issue?

Thank you for looking into this!

rwaldron, 2020-10-29:
> I don’t think the Accelerometer class accepts this option 

Every component constructor accepts a `board` :D

ChristophLabacher, 2020-10-31:
I have not been able to create a short reproducible example – apparently in the depths of our code the direct `this` reference to the board gets lost, making it necessary to pass through the board explicitly. I have not been able to simulate this in a small example – in all my example, the reference to the board was always available and the code didn’t crash.

However, looking through the code for the `Accelerometer` class, I was able to figure out what caused the crash and could confirm this by changing it directly in my local copy of J5. With this fix, the crash does not happen in our app anymore:

I noticed in the stack trace that the crash came from calling `Board.Component.call()`. This is called once in the constructor of `Accelerometer` ([lines 919 – 921](https://github.com/rwaldron/johnny-five/blob/master/lib/accelerometer.js#L919-L921)). At this point it does not crash. It is then called once again by the `Expander` ([lines 1951 – 1956](https://github.com/rwaldron/johnny-five/blob/master/lib/expander.js#L1951-L1956)), which is where it causes the crash.

The reason that it works once but doesn’t work the second time, is that when the `Accelerometer` instantiates the `Expander`, it passes only very few of its options, leaving out the reference to `board` ([lines 694 – 698](https://github.com/rwaldron/johnny-five/blob/master/lib/accelerometer.js#L694-L698)):

```js
        state.expander = Expander.get({
          address,
          controller: this.controller,
          bus: this.bus,
        });
```

Going forward, the `Expander`’s options do not include `board`, and so it cannot be used when the `Expander` calls `Board.Component.call()` with its options.

By including the reference to the `board` in the options for the `Expander` it is passed down as necessary, and the crash does not happen:

```js
        state.expander = Expander.get({
          address,
          controller: this.controller,
          bus: this.bus,
          board: options.board
        });
```

@rwaldron  If you agree with this fix I’d be happy to create a PR.

ChristophLabacher, 2020-11-24:
@rwaldron Is there anything I can contribute to move this along?

closed 2021-07-25 by ChristophLabacher