rwaldron/johnny-five #1713: Sensor on A2 stops Button on 2 from working

tysseng, 2020-11-09:
**Board:** Arduino Uno
**Hardware:** 1 button on pin 2, one potentiometer on pin A2
**Version:** 2.0.0

**My expectations:** It should be possible to have a button on digital pin 2 and a potentiometer on analog pin 2
**Actual outcome:** Button on digital pin 2 stops working when initialising sensor on A2. If I initialise the sensor on A2 first, the button works.

**Details:** 
I suspect that there is something wrong on line 61 in sensor.js:

`this.io.pinMode(this.pin, this.mode);`

When initializing a sensor with pin "A2" or "16", this is translated to "2" before reaching this line. Is it possible that the line above actually sets mode on (digital) pin 2 instead of analog pin 2? 

When I have a button on "2", this stops working if I add a sensor on pin "A2". If I add the button AFTER the sensor in the code, everything works fine. Example:

**Steps to reproduce:**

Non working:
`var five = require("johnny-five"), button, led, pot;
new five.Board().on("ready", function() {
  led = new five.Led(7);
  
  button = new five.Button({ pin: 2, isPullup: true });
  button.on("down", () => led.on());
  button.on("up", () => led.off());

  pot = new five.Sensor("A2");
});`

Working:
`var five = require("johnny-five"), button, led, pot;
new five.Board().on("ready", function() {
  led = new five.Led(7);
  pot = new five.Sensor("A2");
  
  button = new five.Button({ pin: 2, isPullup: true });
  button.on("down", () => led.on());
  button.on("up", () => led.off());
});`

Removing the pinMode on  line 61 in sensor.js also fixes the issue.

rwaldron, 2020-11-15:
It's possible that what you're saying accurately describes the issue, but that shouldn't happen. There is a flow through Board.Component and Board.Options that should normalize pins to whatever the IO plugin defines. In this case "A*" _should_ be normalized to an index that matches analog pin 2. It's possible that logic has gone missing. I'll take a look this week.

rwaldron, 2020-11-17:
Looks like there may be a bug in Firmata.js. IIRC, `pinMode(pin, mode)` should reconcile `pin` to the correct index  based on `mode`.

rwaldron, 2020-11-17:
Looks like Firmata has always done it the way it currently is. It's possible this bug has existed forever and people either didn't hit it or did and just worked around it without reporting it.

rwaldron, 2020-11-17:
@tysseng please try the following: 

```
npm uninstall johnny-five; npm install johnny-five;
```

Then re-run your program and report back whether the problem is resolved. Thanks!!

closed 2022-04-17 by stale[bot]