rwaldron/johnny-five #1716: something with serialport

envythegreat, 2020-11-30:
hello i'm traying to install johnny-five and i keep getting this error 
`warning Error running install script for optional dependency: "C:\\Users\\Envy-Greed\\Desktop\\jn5\\node_modules\\@serialport\\bindings: Command failed.
Exit code: 1
Command: prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
Arguments:
Directory: C:\\Users\\Envy-Greed\\Desktop\\jn5\\node_modules\\@serialport\\bindings
Output:
prebuild-install WARN install No prebuilt binaries found (target=14.15.1 runtime=node arch=x64 libc= platform=win32)

C:\\Users\\Envy-Greed\\Desktop\\jn5\\node_modules\\@serialport\\bindings>if not defined npm_config_node_gyp (node \"C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\node-gyp-bin\\\\..\\..\\node_modules\\node-gyp\\bin\\node-gyp.js\" rebuild )  else (node \"\" rebuild )
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@14.15.1 | win32 | x64
gyp info find Python using Python version 3.9.0 found at \"C:\\Python39\\python.exe\"
gyp ERR! find VS
gyp ERR! find VS msvs_version was set from command line or npm config
gyp ERR! find VS - looking for Visual Studio version 2017
gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
gyp ERR! find VS checking VS2019 (16.8.30717.126) found at:
gyp ERR! find VS \"C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\"
gyp ERR! find VS - found \"Visual Studio C++ core features\"
gyp ERR! find VS - found VC++ toolset: v142
gyp ERR! find VS - found Windows SDK: 10.0.18362.0
gyp ERR! find VS - msvs_version does not match this version
gyp ERR! find VS checking VS2019 (16.7.30413.136) found at:
gyp ERR! find VS \"C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\"
gyp ERR! find VS - found \"Visual Studio C++ core features\"
gyp ERR! find VS - found VC++ toolset: v142
gyp ERR! find VS - missing any Windows SDK
gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
gyp ERR! find VS looking for Visual Studio 2015
gyp ERR! find VS - not found
gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
gyp ERR! find VS
gyp ERR! find VS valid versions for msvs_version:
gyp ERR! find VS - \"2019\"
gyp ERR! find VS - \"C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\"
gyp ERR! find VS
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the \"Desktop development with C++\" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
gyp ERR! find VS
gyp ERR! configure error
gyp ERR! stack Error: Could not find any Visual Studio installation to use
gyp ERR! stack     at VisualStudioFinder.fail (C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\find-visualstudio.js:121:47)
gyp ERR! stack     at C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\find-visualstudio.js:74:16
gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\find-visualstudio.js:351:14)
gyp ERR! stack     at C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\find-visualstudio.js:70:14
gyp ERR! stack     at C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\find-visualstudio.js:372:16
gyp ERR! stack     at C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\util.js:54:7
gyp ERR! stack     at C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\lib\\util.js:33:16
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:315:5)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at maybeClose (internal/child_process.js:1048:16)
gyp ERR! System Windows_NT 10.0.18363
gyp ERR! command \"C:\\\\Program Files\\\\nodejs\\\\node.exe\" \"C:\\\\Program Files\\\\nodejs\\\\node_modules\\\\npm\\\\node_modules\\\\node-gyp\\\\bin\\\\node-gyp.js\" \"rebuild\"
gyp ERR! cwd C:\\Users\\Envy-Greed\\Desktop\\jn5\\node_modules\\@serialport\\bindings
gyp ERR! node -v v14.15.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok"
Done in 19.81s.`

dtex, 2020-11-30:
Hi @envythegreat,

You are correct, this is related to Serialport, not Johnny-Five. There is no prebuilt binary for this version of node so it is trying to build using local tools on your machine. Unfortunately, it's not finding what it needs...

````bash
gyp ERR! find VS **************************************************************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **************************************************************
````

Based on some of the issues being opened over on the Serialport repo, there may be additional issues beyond just having VS installed.

The easy solution is probably to just downgrade your node version. I'm sure a prebuilt binary for your platform/version will be along soon, but I couldn't say when.

Since you've opened an issue over on the Serialport repo, I'm going to go ahead and close this one.

closed 2020-11-30 by dtex

envythegreat, 2020-11-30:
Yeah i downgrade my nodejs to v12 and everything worked perfectly Thank u