rwaldron/johnny-five #1717: Typescript: Add definitions

gpfister, 2020-12-02:
Hi,

Thanks for the awesome project. Would it be possible to have the type definition for those of us who use Typescript ?

Many thanks in advance.
KR,
Greg.

HipsterBrown, 2020-12-03:
Community supported types are available under the DefinitelyTyped project: https://www.npmjs.com/package/@types/johnny-five

`npm install -D @types/johnny-five` should provide the definitions for your project.

gpfister, 2020-12-05:
Hi, unfortunately this is outdated (or incomplete, either way). I understand that the language of choice it javascript, however exposing the API to typescript addict (such as myself) would great. I have started adding the support (on a forked project), right now I'm focusing the API I would use, but if I can get to cover the full documented API, I would gladly submit the pull request.

dtex, 2020-12-12:
Are you planning on submitting a PR to the DefinitelyTyped project or J5?

gpfister, 2021-03-30:
I'm sorry, I'm not going to submit anything back. I have decided to avoid dependencies to community maintained code for everything hardware related, I cannot afford to depend on it for my app. (it doesn't however mean that the work done for this package is rubish, but for production I need to remove some exposure to certain risks on the hardware side).

closed 2021-03-30 by gpfister