rwaldron/johnny-five #1718: My Board cant connect Please help

legoman849, 2020-12-12:
here is my code 

var five = require('johnny-five');
var board = new five.Board();
  
board.on('ready', function(){
    var heart = [
    "01100110",
    "10011001",
    "10000001",
    "10000001",
    "01000010",
    "00100100",
    "00011000",
    "00000000"
  ];

  matrix = new five.Led.Matrix({
    addresses: [0x70],
    controller: "HT16K33",
    rotation: 1});
  
  matrix.clear();
  matrix.draw(heart);

});

and here is the issue

1607790932351 Available COM3
1607790932363 Connected COM3
1607790942364 Device or Firmware Error A timeout occurred while connecting to the Board.

Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.
node:events:367
    throw err; // Unhandled 'error' event
    ^

Error [ERR_UNHANDLED_ERROR]: Unhandled error. ({
  type: 'error',
  timestamp: 1607790942364,
  class: 'Device or Firmware Error',
  message: 'A timeout occurred while connecting to the Board. \n' +
    '\n' +
    "Please check that you've properly flashed the board with the correct firmware.\n" +
    'See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n' +
    '\n' +
    "If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.",
  data: null
})
    at new NodeError (node:internal/errors:278:15)
    at Board.emit (node:events:365:17)
    at Board.EventEmitter.emit (node:domain:470:12)
    at Board.log (C:\Users\benal\node_modules\johnny-five\lib\board.js:642:8)
    at Board.<computed> [as error] (C:\Users\benal\node_modules\johnny-five\lib\board.js:653:14)
    at Timeout._onTimeout (C:\Users\benal\node_modules\johnny-five\lib\board.js:384:14)
    at listOnTimeout (node:internal/timers:556:17)
    at processTimers (node:internal/timers:499:7) {
  code: 'ERR_UNHANDLED_ERROR',
  context: {
    type: 'error',
    timestamp: 1607790942364,
    class: 'Device or Firmware Error',
    message: 'A timeout occurred while connecting to the Board. \n' +
      '\n' +
      "Please check that you've properly flashed the board with the correct firmware.\n" +
      'See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting\n' +
      '\n' +
      "If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.",
    data: null
  }
}

legoman849, 2020-12-12:
im using a arduino uno and node.js

rwaldron, 2020-12-12:
I apologize if this seems like a silly question, but I would be remiss if I didn't ask: have you made sure that Firmata is installed on the Uno?

legoman849, 2020-12-13:
ok now its showing this issue

C:\Users\benal>node heart.js
C:\Users\benal\node_modules\bindings\bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → C:\Users\benal\node_modules\@serialport\bindings\build\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\build\Debug\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\build\Release\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\out\Debug\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\Debug\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\out\Release\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\Release\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\build\default\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\compiled\14.15.1\win32\x64\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\addon-build\release\install-root\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\addon-build\debug\install-root\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\addon-build\default\install-root\bindings.node
 → C:\Users\benal\node_modules\@serialport\bindings\lib\binding\node-v83-win32-x64\bindings.node
    at bindings (C:\Users\benal\node_modules\bindings\bindings.js:126:9)
    at Object.<anonymous> (C:\Users\benal\node_modules\@serialport\bindings\lib\win32.js:1:36)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (C:\Users\benal\node_modules\@serialport\bindings\lib\index.js:6:22)
    at Module._compile (internal/modules/cjs/loader.js:1063:30) {
  tries: [
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\build\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\build\\Debug\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\build\\Release\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\out\\Debug\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\Debug\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\out\\Release\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\Release\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\build\\default\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\compiled\\14.15.1\\win32\\x64\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\addon-build\\release\\install-root\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\addon-build\\debug\\install-root\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\addon-build\\default\\install-root\\bindings.node',
    'C:\\Users\\benal\\node_modules\\@serialport\\bindings\\lib\\binding\\node-v83-win32-x64\\bindings.node'
  ]
}

legoman849, 2020-12-13:
and im using the same code

legoman849, 2020-12-13:
ok i solved it thank you

closed 2020-12-13 by legoman849