rwaldron/johnny-five #1720: Is it possible to combine this project with a esp8266 and express to make a http server

ViniDalvino, 2020-12-16:


dtex, 2020-12-16:
Here is a nice article that can get you up and running with J5 and an esp8266 board.

https://boneskull.com/how-to-use-an-esp8266-with-johnny-five/

Once you have that, you are free to add Express or whatever else you like to your project.

ViniDalvino, 2020-12-16:
let's say I am going to use a arduino to make a internet connection, and another one to use my express server. would it be the right thing to do? @dtex

dtex, 2020-12-16:
Johnny-Five and Express depend on node. Node can't run on an esp8266. The way we work with esp8266 is to run a program called firmata on them. Firmata is a client that listens for instructions from a host server. The host server runs node, Express, and Johnny-Five and communicates wirelessly with the esp8266.

If you want to run everything in one small package, you'll need a single-board computer (SBC) that can run node (and an operating system). The most popular option for that is the Raspberry Pi.

closed 2020-12-16 by ViniDalvino