rwaldron/johnny-five #1721: dht11 does not return any values

TiboriusDev, 2021-01-11:
I have a dht11 with 3pins connected directly to the "uno" board and I get no values back.
Assigned pins are pin2 / 5v / GND.
I also tried it with pin A0 but the same result.

Here is a picture of the dht11 sensor
https://5.imimg.com/data5/QU/AU/MY-9380557/dht-11-temperature-humidity-sensor-module-500x500.jpg

Do I have to use the 2 board or am I doing something else wrong?

```
var SerialPort = require("serialport").SerialPort;
var five = require("johnny-five");
var board = new five.Board({ port: "COM3"});

board.on("ready", function() {
    var multi = new five.Multi({
        controller: "DHT11_I2C_NANO_BACKPACK",
        pin: 2
    });
    multi.on("change", function(){
        console.log("Thermometer");
        console.log("  celsius           : ", this.thermometer.celsius);
        console.log("  fahrenheit        : ", this.thermometer.fahrenheit);
        console.log("  kelvin            : ", this.thermometer.kelvin);
        console.log("--------------------------------------");
        console.log("Hygrometer");
        console.log("  relative humidity : ", this.hygrometer.relativeHumidity);
        console.log("--------------------------------------");
    });
});
```

Output
```
>> Thermometer
  celsius           :  0
  fahrenheit        :  32
  kelvin            :  273.15
--------------------------------------
Hygrometer
  relative humidity :  0
--------------------------------------
```

dtex, 2021-01-11:
The DHT11 depends on features that are not supported by firmata, the software that Johnny-Five uses to communicate with an Arduino. To get around that, you can use a "backpack" which is essentially a relay between the Arduino and the DHT11. [This example](http://johnny-five.io/api/thermometer/#dht11i2cnanobackpack) shows how that is wired up.

TiboriusDev, 2021-01-11:
thanks for the quick help. That brings me then further.

closed 2021-01-11 by TiboriusDev