rwaldron/johnny-five #1723: Raspberry Pi Keypad 4x4 I2C Nano Backpack leads to a crash

AlexKutschera, 2021-03-03:
Hi there,
I would like to use a 4x4 Keypad with an Nano Backpack. I installed the firmware on the Arduino Nano and implemented the example as shown in the documentation. Everything else works fine, I'm also using an I2C LCD, but with they keypad it keeps crashing and I get this error message
```
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: EREMOTEIO: remote I/O error, read
Emitted 'error' event on Board instance at:
    at J5IO.<anonymous> (/var/smart-treehouse/node_modules/johnny-five/lib/board.js:341:39)
    at J5IO.emit (events.js:315:20)
    at J5IO.EventEmitter.emit (domain.js:467:12)
    at readCB (/var/smart-treehouse/node_modules/j5-io/dist/managers/i2c.js:70:35)
    at /var/smart-treehouse/node_modules/raspi-i2c/dist/index.js:72:17
    at FSReqCallback.wrapper [as oncomplete] (fs.js:539:5) {
  errno: -121,
  code: 'EREMOTEIO',
  syscall: 'read'
}
```
my code is the example from the documentation, but I'm using it on an Raspberry Pi instead of an Arduino, could this be a Problem or do I have to provide the I2C Address or something?
```
const board = new five.Board({
    io: new RaspiIO()
});

board.on("ready", () => {
    DisplayManager.init(); // intitializes I2C LCD
    const keypad = new five.Keypad({
        controller: "4X4_I2C_NANO_BACKPACK"
    });
    ["change", "press", "hold", "release"].forEach(function (eventType) {
        keypad.on(eventType, function (event) {
            console.log("Event: %s, Target: %s", eventType, event.which);
        });
    });
    ...
})
```
I'm using johnny-five 2.0.0 and raspi-io 11.0.0 with node14.16.0

rwaldron, 2021-03-04:
Looks like the error is coming from inside raspi-io

AlexKutschera, 2021-03-04:
for me it feels more like there is some issue with the firmware or any missing parameters I would need to provide as i2cdetect does show the Nano but the issue comes from not getting a callback. At least as far as I came digging into the source code.
```
$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- 0d -- -- // Nano
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- 27 -- -- -- -- -- -- -- -- // LCD
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
/var/smart-treehouse/node_modules/raspi-i2c/dist/index.js
```
function createReadBufferCallback(suppliedCallback) {
    return (err, resultOrBytesRead, result) => {
        if (suppliedCallback) {
            if (err) {
                suppliedCallback(err, null); // this line throws the error
            }
            else if (typeof result !== 'undefined') {
                suppliedCallback(null, result);
            }
            else if (Buffer.isBuffer(resultOrBytesRead)) {
                suppliedCallback(null, resultOrBytesRead);
            }
        }
    };
}
```

AlexKutschera, 2021-03-04:
found the issue, I had to provide the specific i2c address of the Nano for it to work. Even though it uses one address of the four defined for the controller it crashes when querying unused I2C Ports. Would love to see this mentioned in the documentation.