rwaldron/johnny-five #1727: Proximity parameter 'freq' does not exist in ProximityOption

lucas-sg, 2021-05-05:
**Board:** Arduino Mega 2560
**Hardware:** HC-SR04 proximity sensor
**Johnny-Five version:** 2.0.0
**Error:**
```
Argument of type '{ controller: string; pin: number; freq: number; }' is not assignable to parameter of type 'number | ProximityOption'.
Object literal may only specify known properties, and **'freq' does not exist in type 'ProximityOption**'.
```
**Project technologies:**
- Node v14.16.0
- Electron v11.2.0
- Typescript

I think the Proximity API documentation might be outdated, particularly in the [_Parameters_](http://johnny-five.io/api/proximity/#parameters) section. It says that it takes an _options_ object with two required parameters and a third (**_freq_**) which is optional. I was able to instantiate a Proximity object with the first two parameters, but when I tried to change the data events' frequency, supposedly given in ms by the **_freq_** parameter, to make the reading slower, I got the error stated above and the following pop-up in VSCode:

[image]

These are the contents of my _tsconfig.json_:
```
{
  "compilerOptions": {
    "target": "es6",
    "module": "commonjs",
    "lib": ["dom", "es2015", "es2016", "es2017", "esnext"],
    "allowJs": true,
    "jsx": "react",
    "sourceMap": true,
    "outDir": "./dist",
    "strict": true,
    "esModuleInterop": true
  }
}
```

The weird thing is that when I clicked on _Debug anyway_, I could actually see the reading frequency getting lower even though I saw an error report in my terminal.

When I hit `F12` in the _Proximity_ object to see the source code, I see the following:
```
export declare interface ProximityOption {
    pin: number | string;
    controller: string;
}

export declare interface ProximityData {
    cm: number;
    in: number;
}

export declare class Proximity {
    constructor(option: number | ProximityOption);
    on(event: string, cb: () => void): this;
    on(event: "data", cb: (data: ProximityData) => void): this;
    on(event: "change", cb: () => void): this;
}
```

I already looked for previous issues on this repo to see if someone had had problems with the _**freq**_ parameter, but I didn't find anything aside from #610 which was closed due to lack of activity but which only suggested that the _freq_ parameter be changed to _period_ instead.

dtex, 2021-05-05:
The `freq` param is a valid property of the options object in Johnny-Five. It looks like [Definitely Typed](https://github.com/DefinitelyTyped/DefinitelyTyped) is missing that in their interface description.

lucas-sg, 2021-05-06:
Thank you @dtex! I will try to make a pull request over there.

lucas-sg, 2021-05-16:
PR done DefinitelyTyped/DefinitelyTyped#52802 :) The `freq` param should now be a usable property with Typescript as well.

closed 2021-05-16 by lucas-sg