rwaldron/johnny-five #1730: How to disable Proximity sensor

lucas-sg, 2021-06-22:
Hello, 
I want to stop the Proximity object from reading data but there seems to be no function to do so. I saw on the [Sensor API](http://johnny-five.io/api/sensor/#api) that I should be able to disable a Sensor (I'm not sure if my HCSR-04 proximity sensor counts as a Sensor object), yet when I check out the implementation (using Typescript) I don't see such a method.

Am I missing something or is the implementation missing?

dtex, 2021-06-23:
Hi @lucas-sg,

The HCSR-04 does not inherit the sensor class and does not have a `disable()` method or an equivalent. At first blush, it looks like it may be quite a chore to add one since there are so many different flavors of proximity device.

lucas-sg, 2021-06-23:
Hi @dtex, thanks for the quick answer :) Sucks to hear that, though.

This might be out of the issue's scope, but do you know how I could make the callback function that is bound to the _"data"_ event change based on an external flag? I'm currently working with React and I would like to process the data until I say so with a stop button. The problem is that the callback function that is passed as an argument to the _.on("data", ...)_ method of the Proximity object is not able to listen to React's state, and change accordingly. I know that the Proximity object would still be listening to the "data" event, but at least I would be able to ignore the data after a certain point.

Thanks in advance.

dtex, 2021-06-23:
Hmm, thinking about it a bit more, you may be able to interface directly with the emitter. Proximity extends withinable which extends node's event emitter.

Something like this might work to disable the emitter.

```js
myProximity.removeAllListeners("data");
```
Then you could have a listener on react's state that toggles the proximity event emitter.

lucas-sg, 2021-06-25:
The problem that I have with this is that I'm using Typescript and it won't recognize that Proximity is extending Withinable. When I look into the index.d.ts file under node_modules/@types/johnny-five I get no declaration or reference of Withinable at all. Here's what the definition of Proximity looks like:
```
export declare class Proximity {
    constructor(option: number | ProximityOption);
    on(event: string, cb: () => void): this;
    on(event: "data", cb: (data: ProximityData) => void): this;
    on(event: "change", cb: () => void): this;
}

export declare interface ProximityOption {
    pin: number | string;
    controller: string;
    freq?: number;
}

export declare interface ProximityData {
    cm: number;
    in: number;
}
```

dtex, 2021-06-25:
That's coming from Definitely Typed correct? Definitely Typed needs to be updated to add the withinable class and note where it is inherited.

lucas-sg, 2021-06-25:
Indeed that's from Definetely Typed. Do you think it would be too complicated to make a fork of DefinetelyTyped and add the implementation myself? Or do you have any tips of where/how to start with that?

dtex, 2021-06-25:
I think Definitely Typed is not a viable project if the people who use it don't contribute back when it needs an update. I'm not a Typescripter, but looking at the repo changes it looks pretty straightforward. I would strongly encourage you to take a look at their [How can I contribute?](https://github.com/DefinitelyTyped/DefinitelyTyped#how-can-i-contribute) guide. They also give some alternative methods for extending existing types or overriding them altogether.

closed 2021-07-07 by lucas-sg