rwaldron/johnny-five #1731: LCD Turns on but nothing happens

vimark1, 2021-06-25:
I have a LCD board `1602A` https://funduino.de/DL/1602LCD.pdf

I have it connected to my Arduino uno in  the approach described in http://johnny-five.io/api/lcd/#i2c-pcf8574-generic

When I run all looks good.
[image]

But the LCD does will only stay on this state
[image]

What am I doing wrong?

dtex, 2021-06-25:
Have you checked the contrast control potentiometer?

Can you show us the wiring? Specifically SDA and SCL pins at both ends (power looks fine)

vimark1, 2021-06-25:
@dtex thank you for your reply.  I'm actually using this wiring. https://create.arduino.cc/projecthub/arduino_uno_guy/i2c-liquid-crystal-displays-5b806c
But even the arduino example doesn't work, so I'm thinking there's an issue with this lcd board.

vimark1, 2021-06-25:
> Have you checked the contrast control potentiometer?

thank you. that looks better now, will try running something again
[image]

vimark1, 2021-06-25:
[image]
the arduino example is working.. thank you very much, will close this issue

closed 2021-06-25 by vimark1

iamisti, 2022-11-26:
would be good to know what was the issue, because now i'm having the same problem

iamisti, 2022-11-26:
you are my only hope now @vimark1 
I've tried everything but I only get the backlight and no characters are showing.