rwaldron/johnny-five #1735: Extend Jhonny Five with C library

dabit1, 2021-07-29:
Hi everyone!

Is it possible to extend Jhonny Five with a new function written in C? I guess we have to add it to the firmware. But tbh I don't really know how to do it.

The thing is, Jhonny Five is really amazing, but when you can't control some component using it you are forced to go back to C, and that's not cool. So I think we would need an easy way to extend Jhonny Five.

thanks!

dtex, 2021-07-29:
Short answer "yes", but I doubt it's the right path. What are you trying to control?

dabit1, 2021-07-29:
Hi @dtex ! Thanks for replying so quick. 
I want to use a dimmer component for AC which seems not to be supported by Johnny Five.
This is the component: 
https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html

I've been able to make it work using their C code. But I couldn't with JF.

dtex, 2021-07-29:
This is pretty different from anything I've seen. It looks like it's thrashing digital pins to create some form of PWM but it's unclear to me what the frequency is.

I assume you're running on an Arduino? If so, there are a few layers that would have to be modified to include this library... Johnny-Five runs on a host computer and talks to the Arduino which is running some flavor of Firmata. You would need to create a custom version of [firmata](https://github.com/firmata/ConfigurableFirmata) that includes that library. Then you would need to add commands to the firmata protocol that are handled in both firmata.js running on the host computer and firmata running on your Arduino. Then you could create a custom class in Johnny-Five that talks to your Arduino running your own custom flavor of Javascript.

That's a LOT of work. I suspect this thing could be controlled using the LED class that's already in Johnny-Five, but I'm not sure. Since you'd be working with enough current to actually hurt yourself or burn your house down, I'm don't think you should try it.

The easy, safe, and practical solution would be to use a device that just uses plain old PWM. Then you can use the Johnny-Five LED class to control it. Something like: https://www.tindie.com/products/bugrovs2012/pwm-2a-ac-light-dimmer-module-50hz-60hz-120v-240v/ Johnny-Five doesn't need to know it's controlling an AC dimmer. The PWM signal is exactly the same.

It's $20 (US), but the custom code option would be days or weeks of work.

dabit1, 2021-08-03:
@dtex Hi! Sorry for the late response, I've been out of my computer for 5 days.
What this component does is to change the frequency of the light. So with less frequency it looks like with less intensity. It also uses a pin in order to detect the "zero-cross". That's why I think that with JS I won't be able to make it work, I need C.

I understand you solution about using that device & LED class but the thing is that I don't want to be forced to buy that devices. Apart from they are very much expensive, I would prefer to do it using the dimmer I bought.

So I'm afraid the only solution is to modify that custom firmdata. I don't know how to do it but I will work on it :)